shaise/FreeCAD_FastenersWB #1: Problems with nut generation

f3nix, 2015-08-08:
Hi,
After clicking the ISO4032 nut or ISO4033 I'm getting a strange result. Screenshot attached.

Any idea what is going on?

EN1661 nut works as expected...

Cheers,
Mateusz

OS: Debian GNU/Linux unstable (sid)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.16.5334 (Git)
Build type: Release
Branch: master
Hash: 4f6a50ac271b74381a69f411983ba2758082573a
Python version: 2.7.10
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 6.9.0

[image]

shaise, 2015-08-08:
Hi f3nix,
This is very strange, I have tried it both on FreeCad V0.15 and V0.16 (5209 and 5332). And it works well. However I use windows version and I have no Linux to test it. 
Can you open "Report View" and see if you get any error messages? (View -> Views -> Report view)
shai

f3nix, 2015-08-08:
There are no error messages in the report view. The console is also empty.

I do not see this problem with FreeCAD 0.14:

OS: Debian GNU/Linux unstable (sid)
Word size: 64-bit
Version: 0.14.3702 (Git)
Branch: releases/FreeCAD-0-14
Hash: b3368125c63289ec8ce9faec2b2ae4c78d436406
Python version: 2.7.8
Qt version: 4.8.6
Coin version: 4.0.0a
SoQt version: 1.6.0a
OCC version: 6.7.0

Any ideas on how to debug it further?

shaise, 2015-08-08:
Can you try it on a clean V0.15 version? (no other plugins/workbenches)

shaise, 2015-08-08:
Also, please delete all preferences: under ~/.FreeCAD, delete (or backup and delete) user.cfg and system.cfg

f3nix, 2015-08-08:
Hi again shaise,
I've narrowed it down a little bit.

The problem seems to be the OCC version. With OCC 6.7.0 everything works as expected. The OCC 6.9.0 is making trouble...

shaise, 2015-08-08:
Hi f3nix,
I'm working with official freecad releases 15 and 16. both are bundled with OCC 6.8.0, which works fine. I'll see if I can easily get 6.9.0 binaries and test it.

ulrich1a, 2015-08-11:
From the picture, it seems that a boolean operation failed, that makes a hex-part from a round part. The release notes for OCCT 6.9 mentions work on boolean operation. The problem could be, that the flat top ring of the nut touches just the cutting shape. I tried to get rid of complex boolean operations, but this one worked and is very fast. So I could try to replace the one complex boolean operation with two simpler operations. For such a simple part, this approach will take a little bit longer. Will see, when I am having some time.

Ulrich

f3nix, 2015-08-13:
Ok. Thanks Ulrich! :)

I'd like to thank You both ulrich1a and shaise for your great work. :+1:

kkoksvik, 2016-03-15:
I see this problem also on OCC v 6.9.1 on Arch

OS: "Arch Linux"
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.16.6612 (Git)
Build type: Unknown
Branch: master
Hash: 18f0cc634665e15f5191dbfa9ea980a16e25ad26
Python version: 2.7.11
Qt version: 4.8.7
Coin version: 3.1.3
OCC version: 6.9.1

3yan, 2016-05-14:
wrote a fix for it:
http://forum.freecadweb.org/viewtopic.php?f=22&t=6558&p=125042#p125042

problem is that shape for cutting hex head is created in a wrong way.

shaise, 2016-05-14:
Yan's fix checked in.
Mateusz, 
Can you please test and see if now it works for you?

f3nix, 2016-12-31:
Hi.
Sorry for very late reply. Somhow I've missed it.

It works for me now. But I am unable to test with original OCC version this issues has been created with.

I think this issue may be closed.

Thanks guys! :) :+1: 

OS: Debian GNU/Linux unstable (sid)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.17.9453 (Git)
Build type: Debug
Branch: master
Hash: b16fbcd4eeb9a5f9195c500e2693cba25a7916d2
Python version: 2.7.13
Qt version: 5.6.2
Coin version: 4.0.0a
OCC version: 7.1.0

luzpaz, 2017-11-08:
@kkoksvik can you still reproduce after 3yan's fix?

luzpaz, 2018-04-26:
@shaise Not sure but this issue may be fixed

shaise, 2018-04-26:
Is this still an issue?

luzpaz, 2018-09-02:
@f3nix do you mind retesting this ?

f3nix, 2018-09-02:
Hi guys!
I can't reproduce it as before: https://github.com/shaise/FreeCAD_FastenersWB/issues/1#issuecomment-269852980

Now tested with:
OS: Debian GNU/Linux unstable (sid)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.14512 (Git)
Build type: Debug
Branch: master
Hash: 94fa64077b748fd2385a83a82e8affffe2ac06db
Python version: 2.7.15+
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: Polish/Poland (pl_PL)

and latest Fasteners WB.

Thanks! :)

Cheers,
Mateusz

luzpaz, 2018-09-02:
Thanks!
@shaise You can close this issue.

shaise, 2018-09-02:
Closing. Thanks.

closed 2018-09-02 by shaise