shaise/FreeCAD_FastenersWB #2: [Enhancement] Bolt-nut holes

pgilfernandez, 2016-03-21:
Hi,

Since 3D printing is around our lives it is very common to design lots of parts to be printed. For assemblies it is also very common to create all kind of holes or grooves in order to fit bolts and nuts.
As it became a repetitive task, I created a holes/grooves library with a standard play of 0,3mm for easy accommodation of bolt/nuts and easy access for tuning the length of holes.
[Here you have my library for better understanding it](https://github.com/pgilfernandez/FreeCAD_bolts-and-nuts-holes).

Do you think it would be a nice addition to include a parametric set of this "idea" into your workbench?

luzpaz, 2019-08-10:
[image]

shaise, 2019-08-10:
Yes, I agree it will be a very nice addition. I quite busy at the moment, but I will definitely look into it as soon as I have time.

pgilfernandez, 2019-08-12:
With the time I have created a [macro with a GUI](https://github.com/pgilfernandez/FreeCAD_Fastener_holes) to make it even easier to create this kind of "holes" with tolerances.
It has not been well coded as I'm not expert but it works for me. The perfect solution would be to integrate this idea into the Fastener Workbench itself, not only because it would be well coded but because the "holes" would become parametric and would make them much more powerful and easy to use/modify.

luzpaz, 2019-12-10:
https://github.com/pgilfernandez/FreeCAD_Fastener_holes looks appropriate for Fasteners WB.

luzpaz, 2020-03-07:
Still hoping this happens :crossed_fingers:

luzpaz, 2020-05-20:
@shaise  
@ulrich1a  
Maybe consider an ETA on this feature?