shaise/FreeCAD_FastenersWB #3: invalid literal for int() with base 10:unknown

iak92, 2016-04-21:
Hey I downloaded your module and extrcted it to ~/.Freecad/Mod. If I start Freecad it says Fasteners loaded and it's shown in the Menu. But if I try to select it the above Error appears. How can i fix this?

shaise, 2016-04-21:
I understand you use Linux. 
What freecad version do you have?
Unfortunately I do not have a Linux machine, so its hard for me to trace such errors

iak92, 2016-04-21:
FreeCAD 0.16 on an ArchLinux Machine with 4.5.1 Kernel

shaise, 2016-04-21:
Lets see if we can get more info.
Before selecting the workbench, show the report view. From the menu - View
-> Panels -> Report view.
Then go to Edit -> Preferences. Select "General" icon. then select "output
window" tab. Make sure all boxes under "Output" are checked.
Now go ahead and select the fasteners workbench.
See if now you get a more detailed error message in the report view. If so,
copy it and send back.

shai.

iak92, 2016-04-21:
invalid literal for int() with base 10: 'Unknown'
Traceback (most recent call last):
  File "<string>", line 60, in Initialize
  File "~/.FreeCAD/Mod/FreeCAD_FastenersWB-master/FastenerBase.py", line 58, in <module>
    DropButtonSupported = int(FreeCAD.Version()[1]) > 15 and  int(FreeCAD.Version()[2].split()[0]) >= 5165

this is the output I get. It seems that FreeCAD.Version() returns a wrong type.

shaise, 2016-04-21:
> File "", line 60, in Initialize

OK, can you try the following:
Exit freecad if its running.
Using a text editor, edit the file
~/.FreeCAD/Mod/FreeCAD_FastenersWB-master/FastenerBase.py
change the problematic line 58 to:
DropButtonSupported = True
save the file and try running again.
Let me know if it helped.

shai

shaise, 2016-04-21:
OK I think I know the reason. On some freecad builds, the build number
might be missing.
Since this check was for old beta versions of freecad 0.16, I think its
safe to remove it.
I have commited the changes into github, please download the new version
and try it.
The fix i gave you in the prev mail will not work, as line 59 have the same
issue.
Let me know if it worked.

shai

iak92, 2016-04-25:
Hey the new version works 
Thanks

closed 2016-04-25 by iak92