shaise/FreeCAD_FastenersWB #9: Error in CountersunkHoles

MrDorianJames, 2017-06-18:
Not sure if this a problem with FreeCAD, my file or the plugin but i get this error. Not sure what it means but i thhought this would be the appropriate place to start.

> `nurbs configuration file loaded
> Mod pluginloader InitGui.py starting ...
> pluginmanager config file /usr/lib/freecad//Mod/plugins/pluginloaderconfig.yaml
> During initialization the error local variable 'fn2' referenced before assignment occurred in /home/dorian/.FreeCAD/Mod/pluginloader/InitGui.py
> Please look into the log file for further informationnot a freecad file: .STP
> not a freecad file: .STL
> Traceback (most recent call last):
>   File "/home/dorian/.FreeCAD/Mod/fasteners/CountersunkHoles.py", line 557, in claimChildren
>     objs.append(self.Object.baseObject[0])
> <type 'exceptions.TypeError'>: 'NoneType' object has no attribute '__getitem__'
> Traceback (most recent call last):
>   File "/home/dorian/.FreeCAD/Mod/fasteners/CountersunkHoles.py", line 557, in claimChildren
>     objs.append(self.Object.baseObject[0])
> <type 'exceptions.TypeError'>: 'NoneType' object has no attribute '__getitem__'
> Traceback (most recent call last):
>   File "/home/dorian/.FreeCAD/Mod/fasteners/CountersunkHoles.py", line 557, in claimChildren
>     objs.append(self.Object.baseObject[0])
> <type 'exceptions.TypeError'>: 'NoneType' object has no attribute '__getitem__'
> Traceback (most recent call last):
>   File "/home/dorian/.FreeCAD/Mod/fasteners/CountersunkHoles.py", line 557, in claimChildren
>     objs.append(self.Object.baseObject[0])
> <type 'exceptions.TypeError'>: 'NoneType' object has no attribute '__getitem__'
> `

shaise, 2017-06-18:
Can you explain what steps you did to get this error?

MrDorianJames, 2017-06-19:
i made a pocket which consisted of two circular holes on surface. then i changed to the Fastener work bench and select the edges to countersink, I select the counter sink icon and tell it which edges to use by selecting them again. I hit okay and it countersinks them. sorry if this doesn't help. is there a log file I can upload? however, I had to save and close my laptop so i lost the undo states.

shaise, 2017-06-19:
Hi, 
Please provide also version of freecad and operating system. Simply, from freecad menu, goto "Help -> About", Click the "Copy to clipboard" button, and paste it here.

MrDorianJames, 2017-06-19:
OS: Ubuntu 17.04
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit

Version: 0.17.11310 (Git)
Build type: None
Branch: master
Hash: d28bae3f52b3e7cf959e8b9f7bcff264f95db191
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.1.0

shaise, 2017-06-19:
Thank you,

I was able to reproduce the issue. looking into it now.

shaise, 2017-06-19:
Thank you for the note. Bug is now fixed.

closed 2017-06-19 by shaise