shaise/FreeCAD_FastenersWB #10: Attaching fastener to a Body

Redneckita, 2017-10-19:
Hi, just installed this WB: looking good!
Only issue is i can't click on a circle of a feature to attach fasteners. FC shows the fastener in the tree with an error message "Links go out of the allowed scope".

However, should i deselect everything, the fastener spawn at 0,0,0 and i can move it where i need. I lack the offset and flip though; Also i need to move every fastener whenever i modify my model. 
Is this normal? Are there some workarounds?
I installed the WB through FC.

```
OS: elementary OS 0.4.1 Loki
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.17.12412 (Git)
Build type: None
Branch: master
Python version: 2.7.12
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.1.0
Locale: Italian/Italy (it_IT)
```

shaise, 2017-11-08:
This should not happen. Can you post the question at freecad forum? you can attach the problematic file there, and I can test it.

luzpaz, 2018-04-26:
@Redneckita please follow up

Redneckita, 2018-04-26:
@luzpaz the issue has been discussed here: https://forum.freecadweb.org/viewtopic.php?f=22&t=25263

shaise, 2018-05-01:
It is now fixed. Fasteners will now attach to the parent body of any selected feature.

Redneckita, 2018-05-01:
> It is now fixed. Fasteners will now attach to the parent body of any selected feature.

Wonderful! Thanks so much for your work!

luzpaz, 2018-09-02:
@shaise this issue cab be closed as well.

shaise, 2018-09-02:
Closing. Thanks.

closed 2018-09-02 by shaise