shaise/FreeCAD_FastenersWB #20: Threaded solids of revolution as first-clacc citizens

KOLANICH, 2018-02-01:
Threads are very widespread in technic, and there is even a special conventions to mark threads on blueprints without drawing it.

So I guess that all bodies lf revolution in freecad should have a property, making them threaded. This may allow marking threads when making drawings, marking threads without rendering them in 3d (for example it is possible to apply a texture or a shader to show them) and importing threads from OpenSCAD keeping them as metadata not a 3d model.

shaise, 2018-02-01:
Hi Kolanich,
Yes, I guess you are right, but I think right now there is no system in freecad core to treat objects as special entities.

luzpaz, 2018-09-02:
@KOLANICH I recommend opening a feature request on the FreeCAD forums to discuss this further with the devs and community...

luzpaz, 2020-05-20:
@KOLANICH care to follow up and create a forum thread to discuss this?

alexneufeld, 2024-02-25:
This should be closed. It's not relevant to us, as the requested feature would seemingly be part of the FreeCAD core distribution and not an addon module. Therefore there is no work related to the stated problem to be done in this repository.

shaise, 2024-02-25:
agree. closing

closed 2024-02-25 by shaise