shaise/FreeCAD_FastenersWB #22: UNF/UNC threaded rod

reox, 2018-04-10:
Today I was in the need of unified thread standard rods. I patched screwmaker to generate me the 5/16-24 thread:
https://github.com/shaise/FreeCAD_FastenersWB/compare/master...reox:master

I do not know much about inch threads but it seems they are quite the same as the metric ones (except withworth of course...)
So maybe it is worth to integrate more inch threads - but I think it will get quite messy because of all the patching of methods that rely on the metric character...

So maybe someone want to take a look at this - or hint me to a workbench that has threaded rods in all variants (I would like to have trapezoidal as well...)

hhassey, 2018-10-29:
I support @reox  I think we should have threaded rods.

hhassey, 2018-10-29:
@reox  It looks like you can get the result with the Screw Tap!

reox, 2018-10-29:
@hhassey what do you mean by Screw Tap? Unfortunately I have no clue what you mean :/

luzpaz, 2019-06-02:
@hhassey ping

rebeltaz, 2019-09-01:
I, too, would like to see SAE threads available in the threaded rod tool.

oldninja, 2021-12-11:
Any plans to support US standards, UNC/UNF threads? @shaise

alexneufeld, 2024-02-25:
This should be closed. We support inch threaded rod.

shaise, 2024-02-25:
correct. closed as complete

closed 2024-02-25 by shaise