shaise/FreeCAD_FastenersWB #24: ISO7046 Screw Fails to generate the screw

hhassey, 2018-05-18:
OS: Debian GNU/Linux 9.4 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.13721 (Git)
Build type: Release
Branch: master
Hash: afdd30b137cdfd096c61c4aee270025115d0f8d6
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 6.8.0.oce-0.17
Locale: English/UnitedStates (en_US)

Just pressing the button with no part associated. creates this error

Traceback (most recent call last):
  File "/home/orion/.FreeCAD/Mod/Fasteners/FastenersCmd.py", line 119, in execute
    s = screwMaker.createScrew(fp.type, d, l, threadType, True)
  File "/home/orion/.FreeCAD/Mod/Fasteners/screw_maker.py", line 2203, in createScrew
    screw = self.makeIso7046(ST_text, ND_text,l)
  File "/home/orion/.FreeCAD/Mod/Fasteners/screw_maker.py", line 3452, in makeIso7046
    recess, recessShell = self.makeCross_H3(cT, mH, ht)
  File "/home/orion/.FreeCAD/Mod/Fasteners/screw_maker.py", line 5000, in makeCross_H3
    crossFaces.append(crossShell.Faces[i])
<type 'exceptions.IndexError'>: list index out of range

shaise, 2018-08-06:
Does it still happen in the latest freecad and FastenersWorkbench? I can not reproduce the issue.

luzpaz, 2018-09-02:
@hhassey please respond

hhassey, 2018-10-29:
I am sorry, I somehow didn't get notified.

It still happens to me.

I just reinstalled the Fasteners WB again to make sure.

My System:
OS: Debian GNU/Linux 9.5 (stretch)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.15007 (Git)
Build type: Debug
Branch: master
Hash: 70fef3934190aabc61f5e1346483359d5ad21403
Python version: 2.7.13
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 6.8.0.oce-0.17
Locale: English/UnitedStates (en_US)

Traceback (most recent call last):
  File "/home/orion/.FreeCAD/Mod/Fasteners/FastenersCmd.py", line 119, in execute
    s = screwMaker.createScrew(fp.type, d, l, threadType, True)
  File "/home/orion/.FreeCAD/Mod/Fasteners/screw_maker.py", line 2203, in createScrew
    screw = self.makeIso7046(ST_text, ND_text,l)
  File "/home/orion/.FreeCAD/Mod/Fasteners/screw_maker.py", line 3452, in makeIso7046
    recess, recessShell = self.makeCross_H3(cT, mH, ht)
  File "/home/orion/.FreeCAD/Mod/Fasteners/screw_maker.py", line 5000, in makeCross_H3
    crossFaces.append(crossShell.Faces[i])
<type 'exceptions.IndexError'>: list index out of range
Screw: list index out of range

hhassey, 2018-10-29:
After trying to debug, I found I had two folders named fasteners (Fasteners and fasteners) I uninstalled, deleted the still there Fasteners folder and re installed, no issue now! 
\
Thanks for your time

We can close this issue now

closed 2018-10-29 by hhassey