shaise/FreeCAD_FastenersWB #25: ISO10642 & ISO14584 wrong direktion socket pocket @freecad 0.18.13780

Gesiima, 2018-05-25:
[image]

shaise, 2018-05-27:
yes, problem verified. It happens freecad when moved from OCC v7.1 to OCC v7.2
Looking into it

shaise, 2018-08-06:
Sorry about that. I dug into the code, (its not mine) but could not find an easy fix. Do you know if ScrewMaker macro works for these types? If so, I will copy the solution from that. If not, I will try again as soon as I have time.

Gesiima, 2018-09-02:
Tested and works great! Thx für fix

OS: Windows 10
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.14495 (Git)
Build type: Release
Branch: master
Hash: c717d1121d888fe6cfbd8971b5f59311a34b37bc
Python version: 2.7.14
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.2.0
Locale: German/Germany (de_DE)

starcatter, 2022-10-08:
Hi, is this issue back? I just installed what looks like the current version (last change in `readme.md` is `V0.4.15  04 Oct 2022`), and upon adding "hexagon socket countersunk head" screws got the following:

[image]

hexalobular screws on the left look ok, hex socket screws on the right are glitched again

shaise, 2022-10-08:
I will take a look at it, can you say what size and length are the hex screws?

starcatter, 2022-10-09:
[image]

shaise, 2022-10-09:
I can not reproduce this issue with the current freecad version:
[image]
Perhaps you should update your freecad, version 18 is quite old.

starcatter, 2022-10-09:
You are right! It works fine in the current stable version:

[image]

Problem occurs in link branch, installed via snap:
[image]

As well as one built from arch package:
[image]

I also ensured all freecad installations share the same version of the fastener workbench. Looks like issue lies with the link branch.

luzpaz, 2022-10-09:
;Please let realthunder know. Closing ticket as this is related to Link branch

closed 2022-10-09 by luzpaz

shaise, 2022-10-13:
I have merged @realthunder 's solution. However this fix is only for non flat countersunk screws so I wonder if it solves the issue. @starcatter , can you please check if it works now with the link branch?

starcatter, 2022-10-15:
Countersunk hex screw sockets are now modeled correctly: 
[image]

However while playing around I noticed two things I think are new.

1 - set screws are sometimes modeled incorrectly, with the socket shape placed out of alignment:
[image]
Looks like it moved after generating the thread:
[image]

2 - after some operations on fasteners (changing types eg. M8->M6, adding thread) an object named "Shape" appears. This happens randomly / isn't always reproducible, but consistently happens after a few operations.
[image]
[image]

Both might be separate issues, I'll test if either happens in the stable branch weekly build. Here's the model I'm testing on:
[5mm-plate.zip](https://github.com/shaise/FreeCAD_FastenersWB/files/9792098/5mm-plate.zip)

starcatter, 2022-10-15:
The extra shape also appears in the weekly build:
[image]

[code]
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.30492 (Git) AppImage
Build type: Release
Branch: master
Hash: b4578fb3eae842eb510a1bc055d07d369a67ce94
Python 3.10.6, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * fasteners 0.4.15
  * Assembly3 0.11.3
[/code]

shaise, 2022-10-16:
Thanks for the report, working on it.
Can you save a file with a bad set screw and attach it?

shaise, 2022-10-16:
@starcatter 
I think I fixed both issues. (no need for example file) Can you please test?

shai

starcatter, 2022-10-16:
Looks fixed - I can't reproduce any of the issues after updating to the latest version. Thanks!