shaise/FreeCAD_FastenersWB #26: Custom bots/screws

rtm7777, 2018-06-27:
Hello,
Thank you guys for your work!

I have M6x70 ISO4762 screws in my workshop, but it's not available it workbench
is there some way to add it? 
The same for M2.5x12 ISO7380

Thanks!

billy1380, 2018-08-23:
It would be great to have a way to select different pitches/coarseness e.g. M10 can be 0.75 1.0 and 1.25 https://www.newmantools.com/tech/threadmf.htm

luzpaz, 2019-06-02:
@rtm7777 can you please post to https://forum.freecadweb.org/viewforum.php?f=8&sid=853eff68d2a09bfd39fb3508d038af97 to start a discussion regarding this issue? Thanks

pandaero, 2021-01-20:
I would like to re-visit this feature request, it would be very useful for 3D-printing unusually large diameter-pitch combinations that are more printable than fine pitches/threads e.g. M20x5 (pitch) x 60 (length), maybe even adding custom thread angle? I will try to parameterise a generic thread in the meantime

ferdymercury, 2021-04-01:
@pandaero this parameterized freecad file might help you:
https://github.com/shaise/FreeCAD_FastenersWB/issues/67#issuecomment-723028691

alexneufeld, 2024-02-25:
This should be closed. We explicitly  don't support custom fasteners in order to limit the complexity of the workbench. Users who want this functionality (seemingly usually for 3D printing purposes???) can already use FreeCAD to make whatever custom fasteners they want, provided they put in a few minutes of their own effort.

shaise, 2024-02-25:
I tend to agree with that. However perhaps we should add at least a custom threaded rod where you can set a custom diameter and thread pitch. This will make it easy for 3d makers to design their own custom fasteners