shaise/FreeCAD_FastenersWB #30: ISO 14584 Screw

OMTDesign, 2018-09-20:
OS: Ubuntu 16.04.5 LTS
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.14779 (Git)
Build type: Release
Branch: master
Hash: 50ab6e292cc83cada4a5b071531a76f5a2580f60
Python version: 2.7.12
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedKingdom (en_GB)

There seems to be a problem with the ISO 14584 screws. The torx pattern is generated above the screw head, and the screw is generated as a hollow object. I tested the bug using all the available parameters and it appeared every time.

[image]

[image]

shaise, 2018-09-21:
This is an old bug. are you sure you use the latest FastenersWB version?

luzpaz, 2018-10-06:
@OMTDesign ping

OMTDesign, 2018-10-09:
> This is an old bug. are you sure you use the latest FastenersWB version?

It turns out that I couldn't update the Fasteners wrokbench through the Freecad Add-on Manager. I uninstalled the addon, reinstalled FastenersWB and the issue was gone.

> @OMTDesign ping

Sorry for the late reply, my inbox got swamped by other messages.

closed 2018-10-09 by OMTDesign