shaise/FreeCAD_FastenersWB #31: Enhancement; nut and bolts to fit when 3D printed and user adjustments

ldboehm, 2018-09-29:
Currently when 3d printing parts with nuts and bolts, they don't fit together.  Some testing shows the nut need to be scaled upward by 110% when printing, then they fit on the bolt thread.  Standard nuts will fit a bolt properly.

enhancement requests
- Nuts and bolts should work together virtual or printed by default.
- Nice to have would be allow some tuning by the user for thread pitch, thread clearance (size the threads up or down a little bit and head size. (hex nut)
- If you wanted to get fancy; add different head types like, philips, torx, star, etc.

shaise, 2018-09-29:
Nuts and bolts are mechanically accurate on CAD. 3D printers add some inaccuracies that makes it not fit that well.
The following fork lets you tinker the sizes of the thread to make it work for 3D printers:
https://github.com/wieck/FreeCAD_FastenersWB/tree/scaling
you can find a discussion in the following thread:
https://forum.freecadweb.org/viewtopic.php?f=8&t=26159&p=207570&hilit=3d+printed+bolt#p207570
I do not understand you third bullet: you can do pretty much any head type you want with this workbench.

ldboehm, 2018-09-29:
Hi Shai,
Thanks for the response.

3D printing technology and printers has advanced greatly, so saying
that it introduces inaccuracies toward what users are experiencing is
not the correct way to state that.  There are lots of things in this
world that started off on paper that needed tuning to match the real
world.

My printer is very new, and it does an amazing job with the things I
print. Some car manufacturers have announced that they are moving all
parts to be created via 3D printing.  The manufacturing industry is
changing from 3D printing.

I read through the thread you provided and seems their findings are
similar to mine, scaling the part when printing is needed to make it
fit right. The discussion says the bolt was scaled down to .98 and the
nut scaled up to 1.05.  I've seen other discussions that use the 1.08
scale.  In these examples and in the thread, the parts are very
simple. If the part were much more complex, then having to scale the
entire part would affect all aspects of the part and that could
introduce other design issues.  I find the bolt is correct, but the
nut needs to be scaled larger during the print process.  I found that
scaling to 110% gave me the desired outcome and the nut fits snug to
the bolt.  What that means is I had to enlarge the entire nut to make
it fit right.

If I and others have to scale down and/or scale up the part during the
print process to make it work, then I'd argue that there is change to
be made in fasteners workbench design.  If you're not going to
entertain reviewing how these parts match up for printing since
mathematically they line up, then I'd suggest creating some type of
tunable attribute within the workbench so a user can scale the nut,
bolt, or clearance of the thread, so the part can be used as it was
designed without having to scale it when printing.

Thanks
-Larry

shaise, 2018-09-30:
As Said,
You can use the modified workbench (first link above) and play with thread
sizes (it will only affect the thread, not other dimensions)
I have been using that myself and even 3D printed functional M3 (!) bolts
and nuts on my Prusa MK3 3D printer.

shai

ldboehm, 2018-09-30:
My apologies, I missed that.
I am confused though, do I just re-install the workbench, or do I have to
install this branch somehow, and if so how?

ldboehm, 2018-10-03:
I'm running MacOS High Sierra and I can't figure out where FreeCAD is
stored!
Any idea?

shaise, 2018-10-03:
you should try and ask on the freecad forum:
https://forum.freecadweb.org/

shIxx01, 2018-12-29:
Hi,
the fork (link above) can not be found via the addon manager. There is only your FastenersWB. But you should find the fork according to its installation instructions in addon manager. However, manual instalation works and the WB is available in the dropdown list in Freecad. but if I also install your WB now  it is strange. The icon of the fork will be displayed, but your WB (where you can not scale) will load.
So both WB can not coexist.

Please fix this issue so both Fastener WB'scan coexist in the addon manager and also in freecad

shaise, 2019-01-26:
The latest version supports 3D printed bolts and nuts. See 
https://forum.freecadweb.org/viewtopic.php?f=8&t=26159&start=20#p282313

luzpaz, 2022-05-31:
@shaise what's left in this ticket?

shaise, 2022-05-31:
dont know. I think the issue is solved

luzpaz, 2022-05-31:
@ldboehm hi, it's been a minute :wink:  
Do you mind looking over this ticket and checking if it's fully satisfied and report back so we know what is its status?

ldboehm, 2022-05-31:
Feel free to close it.  IMHO adjusting thread size and width should be in the automatically updated Workbench.
Having to download code and overwrite manually isn't the greatest user experience.

shaise, 2022-05-31:
@ldboehm , this feature exists in the formal, updated, Workbench

luzpaz, 2022-05-31:
@shaise thanks for weighing-in. If issues are closed, please consider closing them. Thanks!