shaise/FreeCAD_FastenersWB #33: Fix up the Fasteners FC wiki page

luzpaz, 2018-10-06:
@yorikvanhavre is intent on having all workbenches have some representation on the wiki. The Fasteners WB is at www.freecadweb.org/wiki/Sandbox:Fasteners
Lets move it out of Sandbox and soup it up?

Also I wonder if there is a way for mediawiki to mirror the changelog for Fasteners on Github ?

shaise, 2018-10-06:
As soon as I have time... :)

luzpaz, 2019-05-14:
I've moved the page out of the sandbox http://www.freecadweb.org/wiki/Fasteners_Workbench

luzpaz, 2019-05-14:
@shaise JIC (you probably already know anyway), there are some broken image links on that wiki page.

shaise, 2019-05-14:
Hi,
I did not make this wiki page. How can I help?

luzpaz, 2019-05-14:
Oh if that's the case, I'm not sure then. Perhaps take a look at it and see if there is anything you think people should be aware about ? If you are feeling comfortable with the idea, you could make a youtube clip or a screencast of the workbench in action? Whatever you think.

I think you should also add the https://www.freecadweb.org/wiki/Fasteners_Workbench links somewhere in the README.md as well.

luzpaz, 2020-03-07:
I started to do this recently

shaise, 2020-03-07:
You are definitely my FreeCad Hero!

luzpaz, 2021-03-31:
[image]

Created placeholder pages for all but 2 pages mentioned in the Fasteners Workbench wiki page

luzpaz, 2021-03-31:
correction: placeholder now exist for all pages

shaise, 2021-04-01:
Thanks!!!!
Added a link to the wiki in readme.md

commit 369c13db (2024-03-10): WIP: Rename GUI commands to follow FreeCAD format

FreeCAD have a Help module where using the What's this? command you can click on a command on the toolbar and
the wiki page for it will open.The way it works is: it opens https://wiki.freecad.org/{$Command_name}.

For Fasteners WB pages are "Fasteners_CommandName", currently command names are "FSCommandName" which leads to an
error 404 when fetching Wiki page.

Fix #33
closed 2024-03-10 by shaise