shaise/FreeCAD_FastenersWB #35: Fales to generate screw ISO4762 M4x10

shIxx01, 2018-12-27:
Hallo,
it seems like you can not generate an ISO 4762 screw if you want it to be 10mm in length.
This error appears in the report view:
```
Traceback (most recent call last):
  File "C:\Users\thomas\AppData\Roaming\FreeCAD\Mod\fasteners\FastenersCmd.py", line 119, in execute
    s = screwMaker.createFastener(fp.type, d, l, threadType, True)
  File "C:\Users\thomas\AppData\Roaming\FreeCAD\Mod\fasteners\ScrewMaker.py", line 249, in createFastener
    return self.createScrew(type, diam, len, threadType, shapeOnly)
  File "C:\Users\thomas\AppData\Roaming\FreeCAD\Mod\fasteners\screw_maker.py", line 2197, in createScrew
    screw = self.makeIso4762(ST_text, ND_text,l)
  File "C:\Users\thomas\AppData\Roaming\FreeCAD\Mod\fasteners\screw_maker.py", line 3741, in makeIso4762
    headShell = aWire.revolve(Base.Vector(0.0,0.0,0.0),Base.Vector(0.0,0.0,1.0),360)
<class 'Part.OCCError'>: BRepAdaptor_Curve::No geometry
```

System:
OS: Windows 10
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.17.13541 (Git)
Build type: Release
Branch: releases/FreeCAD-0-17
Hash: 9948ee4f1570df9216862a79705afb367b2c6ffb
Python version: 2.7.14
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.2.0
Locale: German/Germany (de_DE)

shIxx01, 2018-12-27:
Strange, as soon as I set thread to true it workes. Here is the whole report file:
`Traceback (most recent call last):
  File "C:\Users\thoma\AppData\Roaming\FreeCAD\Mod\fasteners\FastenersCmd.py", line 119, in execute
    s = screwMaker.createFastener(fp.type, d, l, threadType, True)
  File "C:\Users\thoma\AppData\Roaming\FreeCAD\Mod\fasteners\ScrewMaker.py", line 249, in createFastener
    return self.createScrew(type, diam, len, threadType, shapeOnly)
  File "C:\Users\thoma\AppData\Roaming\FreeCAD\Mod\fasteners\screw_maker.py", line 2197, in createScrew
    screw = self.makeIso4762(ST_text, ND_text,l)
  File "C:\Users\thoma\AppData\Roaming\FreeCAD\Mod\fasteners\screw_maker.py", line 3741, in makeIso4762
    headShell = aWire.revolve(Base.Vector(0.0,0.0,0.0),Base.Vector(0.0,0.0,1.0),360)
<class 'Part.OCCError'>: BRepAdaptor_Curve::No geometry
Traceback (most recent call last):
  File "C:\Users\thoma\AppData\Roaming\FreeCAD\Mod\fasteners\FastenersCmd.py", line 119, in execute
    s = screwMaker.createFastener(fp.type, d, l, threadType, True)
  File "C:\Users\thoma\AppData\Roaming\FreeCAD\Mod\fasteners\ScrewMaker.py", line 249, in createFastener
    return self.createScrew(type, diam, len, threadType, shapeOnly)
  File "C:\Users\thoma\AppData\Roaming\FreeCAD\Mod\fasteners\screw_maker.py", line 2197, in createScrew
    screw = self.makeIso4762(ST_text, ND_text,l)
  File "C:\Users\thoma\AppData\Roaming\FreeCAD\Mod\fasteners\screw_maker.py", line 3741, in makeIso4762
    headShell = aWire.revolve(Base.Vector(0.0,0.0,0.0),Base.Vector(0.0,0.0,1.0),360)
<class 'Part.OCCError'>: BRepAdaptor_Curve::No geometry
Traceback (most recent call last):
  File "C:\Users\thoma\AppData\Roaming\FreeCAD\Mod\fasteners\FastenersCmd.py", line 119, in execute
    s = screwMaker.createFastener(fp.type, d, l, threadType, True)
  File "C:\Users\thoma\AppData\Roaming\FreeCAD\Mod\fasteners\ScrewMaker.py", line 249, in createFastener
    return self.createScrew(type, diam, len, threadType, shapeOnly)
  File "C:\Users\thoma\AppData\Roaming\FreeCAD\Mod\fasteners\screw_maker.py", line 2197, in createScrew
    screw = self.makeIso4762(ST_text, ND_text,l)
  File "C:\Users\thoma\AppData\Roaming\FreeCAD\Mod\fasteners\screw_maker.py", line 3741, in makeIso4762
    headShell = aWire.revolve(Base.Vector(0.0,0.0,0.0),Base.Vector(0.0,0.0,1.0),360)
<class 'Part.OCCError'>: BRepAdaptor_Curve::No geometry`

shaise, 2018-12-28:
Hi,
Thanks for the catch. I have fixed it, please test and close the issue.

shai

closed 2018-12-29 by shIxx01