shaise/FreeCAD_FastenersWB #36: Does not coexist with the fork

shIxx01, 2019-01-06:
Hi,
the fork (https://github.com/wieck/FreeCAD_FastenersWB/tree/scaling) can not be found via addon manager. There is only your FastenersWB. But you should find the fork according to its installation instructios in addon manager. However, manual instalation works and the WB is available in the dropdown list in Freecad. But if I also install your WB now,  it behaves strange. The icon of the fork will be displayed, but your WB (where you can not scale) will load.
So both WB can not coexist (and yes, they are in different folders).

Please fix this issue so both Fastener WB's can coexist in the addon manager and also in freecad

_Originally posted by @shIxx01 in https://github.com/shaise/FreeCAD_FastenersWB/issues/31#issuecomment-450487174_

luzpaz, 2019-01-25:
@shIxx01 that's because the is the official Fastener WB and @wieck forked it. So it makes sense that it isn't in the addon manager. Perhaps @wieck can open a PR and a discussion on the forums with a merge proposal ?

wieck, 2019-01-27:
My fork is/was experimental, only implementing the thread scaling for a few bolt/nut types as a proof of concept. Job changes kept me from getting back to it.

Good to hear that a new version has this feature. The way I hacked it led to invalid objects that slic3r wouldn't load properly. 

I am going to drop the fork to avoid future confusion.

Thanks for all your work, Jan

shaise, 2019-01-27:
Thank you Jan for your Idea. 
I found it useful when I need to tap holes into models.

shIxx01, 2019-01-27:
Ok Tanks for the feature. I will close this issue

closed 2019-01-27 by shIxx01