shaise/FreeCAD_FastenersWB #39: Installing Latest Fasteners on Appimage errors

luzpaz, 2019-01-25:
Running:
```
OS: Manjaro Linux
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.15671 (Git) AppImage
Build type: Release
Branch: master
Hash: 28cbe2bd8986701bb7404842cf60138957e23ed6
Python version: 3.6.7
Qt version: 5.6.2
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```
After installing latest Fasteners WB and restarting FC  
Switching to Fasteners returns: 
```
invalid syntax (ScrewMaker.py, line 258)
Traceback (most recent call last):
  File "<string>", line 45, in Initialize
  File "/home/beast/.FreeCAD/Mod/fasteners/FSScrewCalc.py", line 141, in <module>
    import ScrewMaker  
```

shaise, 2019-01-25:
can't recreate the error under my system:
OS: Windows 10
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.15656 (Git)
Build type: Release
Branch: master
Hash: a60b6957db228c42a00a1e002a415cbb8309d3c8
Python version: 2.7.14
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.2.0
Locale: English/UnitedStates (en_US)

any Idea where to look?

shaise, 2019-01-25:
Ok, I think I know why. Please try again.

luzpaz, 2019-06-03:
Works :+1:

closed 2019-06-03 by luzpaz