shaise/FreeCAD_FastenersWB #42: Error creating thread in M5 nut

TreidenX, 2019-01-27:
OS: Windows 10
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.17.13541 (Git)
Build type: Release
Branch: releases/FreeCAD-0-17
Hash: 9948ee4f1570df9216862a79705afb367b2c6ffb
Python version: 2.7.14
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.2.0

Error creating thread in M5 nut (ISO-4032, ISO-4033, ISO-4035):

Traceback (most recent call last):
  File "C:\Users\atnwe\AppData\Roaming\FreeCAD\Mod\fasteners\FastenersCmd.py", line 121, in execute
    s = screwMaker.createFastener(fp.type, d, l, threadType, True)
  File "C:\Users\atnwe\AppData\Roaming\FreeCAD\Mod\fasteners\ScrewMaker.py", line 264, in createFastener
    return self.createScrew(type, diam, len, threadType, shapeOnly)
  File "C:\Users\atnwe\AppData\Roaming\FreeCAD\Mod\fasteners\screw_maker.py", line 2230, in createScrew
    screw = self.makeIso4032(ST_text, ND_text)
  File "C:\Users\atnwe\AppData\Roaming\FreeCAD\Mod\fasteners\screw_maker.py", line 4655, in makeIso4032
    nutShell = Part.Shell(nutFaces)
<class 'Part.OCCError'>: Shape is not a shell

Error creating thread in M5 nut (EN-1661):

Traceback (most recent call last):
  File "C:\Users\atnwe\AppData\Roaming\FreeCAD\Mod\fasteners\FastenersCmd.py", line 121, in execute
    s = screwMaker.createFastener(fp.type, d, l, threadType, True)
  File "C:\Users\atnwe\AppData\Roaming\FreeCAD\Mod\fasteners\ScrewMaker.py", line 264, in createFastener
    return self.createScrew(type, diam, len, threadType, shapeOnly)
  File "C:\Users\atnwe\AppData\Roaming\FreeCAD\Mod\fasteners\screw_maker.py", line 2234, in createScrew
    screw = self.makeEN1661(ND_text)
  File "C:\Users\atnwe\AppData\Roaming\FreeCAD\Mod\fasteners\screw_maker.py", line 4825, in makeEN1661
    headShell = Part.Shell(topFaces)
<class 'Part.OCCError'>: Shape is not a shell

At a choice DIN-557, DIN-562, DIN-985 the thread is not created with any size and without errors.

shaise, 2019-01-30:
Yes, I have noticed for sometime. I notified the developer and will implement the fix as soon as he solves it.

pubpsv, 2019-02-23:
Error creating thread in nut M30

gabirx, 2019-04-20:
Error creating thread in nut (M27) - shape is not a shell
Traceback (most recent call last):
  File "C:\Users\user\AppData\Roaming\FreeCAD\Mod\fasteners\FastenersCmd.py", line 121, in execute
    s = screwMaker.createFastener(fp.type, d, l, threadType, True)
  File "C:\Users\user\AppData\Roaming\FreeCAD\Mod\fasteners\ScrewMaker.py", line 264, in createFastener
    return self.createScrew(type, diam, len, threadType, shapeOnly)
  File "C:\Users\alexandr\AppData\Roaming\FreeCAD\Mod\fasteners\screw_maker.py", line 2230, in createScrew
    screw = self.makeIso4032(ST_text, ND_text)
  File "C:\Users\user\AppData\Roaming\FreeCAD\Mod\fasteners\screw_maker.py", line 4655, in makeIso4032
    nutShell = Part.Shell(nutFaces)
<class 'Part.OCCError'>: Shape is not a shell

luzpaz, 2019-05-14:
@shaise any progress?

shaise, 2019-05-14:
@luzpaz This is Ulrich's code. I think it got something to do with OCC.

luzpaz, 2019-05-14:
@shaise can we ping ulrich on this, just so if there needs to be a workaround or an upstream OCC ticket made ?

shaise, 2019-05-14:
We can ping him, I just dont know how.

easyw, 2019-05-14:
@shaise ... I think this is his repo
https://github.com/ulrich1a
than to ping him you can just put @ulrich1a here.
PS @shaise would you please have a look at this [PR 79 @ SheetMetal](https://github.com/shaise/FreeCAD_SheetMetal/pull/79)?

shaise, 2019-05-14:
@easyw, regarding 79, I'm more or less OK with it, But I'm waiting for your approval.

easyw, 2019-05-14:
@shaise I think we can merge it and then apply the fix to drop to old method in case of simple models.
I have also a new pr pending, for improving some unfolding user case as seen at freecad forum.

luzpaz, 2019-08-10:
status on this ?

luzpaz, 2020-02-04:
> status on this ?  

bump

shaise, 2020-02-05:
Still waiting for @ulrich1a to investigate. Don't actually know why the new OCC complains on the thread generation, and also why it only happens for certain M sizes and not others.

Altair31, 2020-06-13:
@ulrich1a did not make a single commit in a year. Do you have any contact ?

shaise, 2020-06-13:
nope!

ulrich1a, 2020-06-29:
I am still alive. I have to admit, that I my focus had moved to other things for a long time. I have been working the last days on the issue with the nuts and found a solution. It is a little bit faster but needs still some tuning. I will upload the current status to my github account.

shaise, 2020-06-29:
great! good to hear from you!

ulrich1a, 2020-06-30:
Added the new code to github: https://github.com/ulrich1a/screw_maker

shaise, 2020-07-01:
great! I will take a look and try to merge it.

shaise, 2020-07-01:
Should be fixed now. Thank you @ulrich1a !

luzpaz, 2020-07-05:
@shaise FYI, if you add `closes #42` to the git summary it will automatically close this ticket.

shaise, 2020-07-05:
Thanks for the tip. However, I think it should be closed by issue owner.

luzpaz, 2021-07-02:
bump @TreidenX