shaise/FreeCAD_FastenersWB #50: Fasteners/nuts/washers in arrays aren't counted in BOM

DDMDesign, 2019-06-03:
Fasteners/nuts/washers in an array aren't counted in BOM.
Fasteners/nuts/washers in a cloned compound aren't counted in BOM.
The same if fasteners/nuts/washers are converted in a structure (Arch/BIM Workbech).

luzpaz, 2019-06-06:
Relevant [forum link](https://forum.freecadweb.org/viewtopic.php?f=24&t=36752#p311727) showing the usecase

luzpaz, 2019-08-10:
@shaise care to weigh in ?

shaise, 2019-08-10:
I'm not all that familiar with FreeCad's arrays. Can you attach a simple FreeCad file with array of bolts so I can try and analyze it's structure?

luzpaz, 2019-12-10:
> I'm not all that familiar with FreeCad's arrays. Can you attach a simple FreeCad file with array of bolts so I can try and analyze it's structure?  

@DDMDesign please follow up

DDMDesign, 2019-12-10:
Sorry, I lost the message. Here attached a simple array with based of a fastner and the BOM generated with fasteners workbench. As you see the quantity is one but in the array is nine.

Thanks, Marco
[Fastner_Array+BOM.zip](https://github.com/shaise/FreeCAD_FastenersWB/files/3945355/Fastner_Array%2BBOM.zip)

DDMDesign, 2019-12-10:
I forgot the FC info. Now I am using 0.19, but it is the same with 0.18 or 0.17...

OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.18516 (Git)
Build type: Release
Branch: master
Hash: 867a09f731926bf75695e8a6f063cbc9c1033b40
Python version: 3.6.8
Qt version: 5.12.1
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: Italian/Italy (it_IT)

DDMDesign, 2019-12-11:
I tried also the App::link of a fastner and BOM of the fastners workbench count only one element.

Here attached the updated file.

Cheers, Marco
[Fastner_Array+BOM.zip](https://github.com/shaise/FreeCAD_FastenersWB/files/3949839/Fastner_Array%2BBOM.zip)

shaise, 2020-02-03:
sorry for the late reply. 
It  should work now, please test.
shai

DDMDesign, 2020-02-04:
Hi Shai, Thanks for looking into it...

I tested some cases:

1) One array 3x3 of a bolt M20: BOM work fine and counts 9
2) Double array 3x3 of a bolt M20: BOM counts 81 bolts, but should be 9+9 =18
3) One array 3x3 of a bolt M20 and the link of the array: BOM counts 81 bolts, but should be 9+9 = 18
4) Links of a bolt M20 (the property "Element count" of the link object is set to 10: BOM counts 1 , but should be 11 (one for the base element plus 10 for the links)

The link object is very useful for large model. You can insert a bolt, create links and use them in the model. If you want to change the diameter you have to change only the property of the base element.

Here attached the 4 files.

I have another question. Can the fasteners BOM count only the bolts/washer/etc for a selection of the model? If a user has more assemblies it can be useful to know fasteners BOM of each assembly...

I hope a was clear.

Cheers, Marco

[Test_BOM_Fasteners.zip](https://github.com/shaise/FreeCAD_FastenersWB/files/4156018/Test_BOM_Fasteners.zip)

shaise, 2020-02-16:
Trying to work on it now.
seems that links are freecad 19 feature, right? I'm using 18 and it is not link aware.

shaise, 2020-02-16:
OK, looks like v0.19 links are buggy (specially when making lins on links or array on links)
but the new version I just commited should work on your examples.

DDMDesign, 2020-02-16:
Thanks Shai!!! All the four examples work fine...
I tried also with three arrays 3x3 and BOM counts 27 elements...
Also the example four (with link objects ) works fine and counts 11 elements...
I will try BOM for simple examples with links of more and different fasteners.

Thanks a lot for this release...Great job....It's very useful...
Can I report to other users this new release in this thread?
https://forum.freecadweb.org/viewtopic.php?f=24&t=36752&sid=2aca9ed558f5c5b297096b286c971d29&start=10
Marco

shaise, 2020-02-17:
Yes of course.

dendres, 2020-07-29:
Thanks for this feature!

Does it work with PathArray?

Steps to reproduce:
 - update 0.19 appimage and all addons and restart
 - new doc
 - add a hex bolt, draw a line
 - select both and click PathArray
 - set count to 20
 - click Generate fasteners bill of material
 - observe Qty = 1

What could be different about PathArray?

luzpaz, 2022-05-31:
Hi @wavexx would you mind taking a swing at this issue?

wavexx, 2022-05-31:
Keep in mind I'm still relatively new at freecad (well, not really, I've been using it for small hacks for years, but with realthunder's fork I got some relatively complex projects working and I'm feeling impressed enough to be ready to jump out of commercial stuff and start contributing).

Reading the report, it seems like it works as intended, except patharray is not behaving as a regular array would. Do I understand this right?

I would personally attempt to fix this in PathArray instead of plugging in exceptions... (mostly to avoid surprises in the long run).

luzpaz, 2022-05-31:
> I would personally attempt to fix this in PathArray instead of plugging in exceptions... (mostly to avoid surprises in the long run). 

That actually sounds like addressing the central problem.

Roy-043, 2022-09-01:
IMO this issue can now be closed.

wavexx, 2022-09-01:
I didn't have time to look at PathArray yet. Was this solved, or maybe we should have a specific issue on patharrays?

Roy-043, 2022-09-01:
The code in https://github.com/shaise/FreeCAD_FastenersWB/pull/168 also handles path arrays.

closed 2022-09-01 by luzpaz