shaise/FreeCAD_FastenersWB #52: [Feature request] Metric size M3.5 for scres and nuts?

gimpox, 2019-12-10:
I love the Fasteners Workbench, unfortunately I'm using M3.5 ISO-7380 screws in my project.
As I can see, there is no way to create such screw. So I'm forced to choose M3 or M4 screws...

The M3.5 size is very common for enthusiast of car/train/plane models, as well as for flying drones.
So it would be nice to have it also for:

- ISO-7380 bolts with flange
- Nuts with and without the nylon locking ring

shaise, 2019-12-10:
The fasteners workbench is based on the macro Screw_Maker by ulrich1a. It does not support m3.5. I can try and add it to the table, but M3.5 is not within the ISO-7380 standard, hence no measurement info:
https://www.fasteners.eu/standards/ISO/7380/

gimpox, 2019-12-10:
As I understand, the ISO standard specify the shape of the thread,that is indicated by the letter "M". The "M"-shape is just one type, there are many other (indicated as "KT", "R", and so on - see http://www.emuge-franken-bg.com/attachments/article/97/15 Gewindetabellen.pdf).
The size of the screws usually saw on the tables are just a "preferred" nominal size. Strictly speaking, there is no theoretical limitation in the diameter size.

On Wikipedia the M3,5 size is listed (among the other) with a pitch of 0.6 mm:
https://en.wikipedia.org/wiki/ISO_metric_screw_thread

shaise, 2019-12-10:
I do not need the iso standard of the thread, i need the iso 7380 standard of m3.5 which contains many parameters regarding the head shape. 
[image]

gimpox, 2019-12-11:
I will investigate...

mendip-defender, 2020-01-23:
Maybe the DIN 912 standard might fill in some blanks,
https://www.torqbolt.com/din-912-dimensions-standards-specifications/

shaise, 2020-01-24:
Not the same head shape, and has no m3.5 standard, regardless.

luzpaz, 2020-05-20:
@gimpox any progress ?

ulrich1a, 2020-06-30:
The right choice of a screw would be ISO 14580. It has a hexalobular socket and (M3.5) as a not recommended size.
Other options: ISO 14583