shaise/FreeCAD_FastenersWB #53: Align Fastener to Hole Programmatically

gbroques, 2020-01-09:
I'd like to call `createFastener` after selecting a circular edge to align a fastener to a hole, but I'm getting the following error.

```
>>> sm.createFastener('ISO4014', 'M24', '120', 'simple')
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "/home/g/snap/freecad/8/.FreeCAD/Mod/fasteners/ScrewMaker.py", line 264, in createFastener
    return self.createScrew(type, diam, len, threadType, shapeOnly)
  File "/home/g/snap/freecad/8/.FreeCAD/Mod/fasteners/screw_maker.py", line 2263, in createScrew
    self.moveScrew(ScrewObj)
  File "/home/g/snap/freecad/8/.FreeCAD/Mod/fasteners/screw_maker.py", line 2280, in moveScrew
    for o in Gui.Selection.getSelectionEx():
NameError: global name 'Gui' is not defined
```
### Screenshot
[image]

Based on the stack-trace, the error is coming from the following line of code:
https://github.com/shaise/FreeCAD_FastenersWB/blob/master/screw_maker.py#L2280

Should this be `FreeCADGui` instead of `Gui`?

I manually changed `Gui` to `FreeCADGui` locally, but when I call `createFastener` it still doesn't align the screw to the selected circular edge or hole.

Is this expected behavior? Any ideas on how to programmatically align the screw to a selected circular edge?

### FreeCAD Version Information
```
OS: Linux
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.16093 (Git)
Build type: Unknown
Branch: master
Hash: 690774c0effe4fd7b8d2b5e2fb2b8c8d145e21ce
Python version: 2.7.12
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.0.0
Locale: English/UnitedStates (en_US)
```

gbroques, 2020-01-09:
It appears the code that correctly attaches the screw to an object (via the GUI) is a function called `FSMoveToObject` in `FastenerBase.py`.

I would've expected the GUI and `createFastener` python function to re-use the same "attach to object" logic.

The `createFastener` function calls `moveScrew` in `screw_maker.py` which doesn't seem to properly work.

I might suggest refactoring `createFastener` to reuse the attach to object logic in the `FSMoveToObject` function.

shaise, 2020-01-09:
can you attache the freecad file with the object you want to attach the fastener to? (zipped)

gbroques, 2020-01-10:
@shaise Yes.

I've attached [attach-screw-to-hole-python-issue-53.zip](https://github.com/shaise/FreeCAD_FastenersWB/files/4043200/attach-screw-to-hole-python-issue-53.zip) which includes:
1. The freecad file with the object I want to attach the fastener to `8-hole-tubing-cube.FCStd`
2. I've also included the individual object in `8 hole tubing.fcstd`

Thank you for your fast response.

gbroques, 2020-01-10:
After changing `Gui` to `FreeCADGui` and uncommenting your print debug statements, I get this in the console:
```
In Move Screw: <Part::PartFeature>
Selektionen: 1
object: <Edge object at 0x53ec220>
The Object is a Curve!
Got Axis1: Vector (-1.0, 0.0, 0.0)
Winkel = 90.0
Normalvektor: Vector (-0.0, -0.7071067811865475, 0.0)
pl mit Rot: Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,-90,-0)]
die Position     Placement [Pos=(0,0,0), Yaw-Pitch-Roll=(0,0,0)]
die rot. Position: Placement [Pos=(-50.8,0,50.8), Yaw-Pitch-Roll=(0,-90,0)]
```

The screw's position isn't updated programmatically.

However, manually updating the screws position in the GUI to the calculated position of **[Pos=(-50.8,0,50.8), Yaw-Pitch-Roll=(0,-90,0)]** seems to be correct. (see below image)

[image]

shaise, 2020-01-13:
Ok, I understand now. You are actually using the original 'moveScrew' used by the original screw_maker macro that I adapted into the workbench.
Instead you should do the following:

f = sm.createFastener('ISO4014', 'M24', '120', 'simple', True)
Part.show(f)
You can now move the part into the holw by using the movetoobject function, but you have to supply the selected feature you want to attach to by getting the gui selection

shai

gbroques, 2020-01-15:
Thanks @shaise.

The `Part.show` didn't seem to be necessary.

The following (incomplete) code snippet worked to attach a bolt to a hole, and nut on the other end:
```python
    # Select hole I want to attach the bolt to
    FreeCADGui.Selection.addSelection(tube, 'Edge71')

    # Simulate clicking ISO 4014 Hex head bolt button in GUI
    FreeCADGui.runCommand('FSISO4014')

    # Update bolt properties and recompute for proper rendering
    bolt = FreeCAD.ActiveDocument.Screw
    bolt.length = '220'
    bolt.invert = True
    FreeCAD.ActiveDocument.recompute()

    # Create nut
    sm = ScrewMaker.Instance()
    nut = sm.createFastener('ISO4032', '(M33)', 0, 'simple')

    # Attach nut to bottom of bolt on the other end of hole
    bottom_bolt_edge = bolt.Shape.Edges[40]
    inverted = True
    offset = 1
    move_to_object = ScrewMaker.FastenerBase.FSMoveToObject
    move_to_object(nut, bottom_bolt_edge, not inverted, offset)
```

---

Do you feel it'd be valuable to document `ScrewMaker.FastenerBase.FSMoveToObject` in the **README**?

I'd be happy to submit a **PR** if it's welcomed.

shaise, 2020-01-15:
The Part.show IS needed when you add the 'True' at the end of the CreateFastener command. You should use the True option, this way avoiding the 'Gui' issue.
I do not think the command should be documented in the readme as the readme is for workbench users and not programmers.
shai

gbroques, 2020-01-16:
> The Part.show IS needed when you add the 'True' at the end of the CreateFastener command. You should use the True option, this way avoiding the 'Gui' issue.

Thanks shai. All this makes a lot more sense now.

I'll go ahead and close out this issue.

I appreciate your help and creating this project for the FreeCAD community.

closed 2020-01-16 by gbroques