shaise/FreeCAD_FastenersWB #54: Document createFastener Python API

gbroques, 2020-01-11:
It'd be helpful to document the valid types and sizes that can be passed into `createFastener`.

Perhaps under the [Adding Fasteners in Python](https://github.com/shaise/FreeCAD_FastenersWB#adding-fasteners-via-python) section in the **README**.

The below is a work in progress MarkDown table to document the `createFastener` **API**.

|Name|Type|Diameters|
|----|----|---------|
|Screw|ISO4017||
|Screw|ISO4014||
|Screw|EN1662||
|Screw|EN1665||
|Screw|ISO2009||
|Screw|ISO2010||
|Screw|ISO4762||
|Screw|ISO10642||
|Screw|ISO1207||
|Screw|ISO1580||
|Screw|ISO7045||
|Screw|ISO7046||
|Screw|ISO7047||
|Screw|ISO7048||
|Screw|DIN967||
|Screw|ISO7380-1||
|Screw|ISO7380-2||
|Screw|ISO14579||
|Screw|ISO14580||
|Screw|ISO14582||
|Screw|ISO14583||
|Screw|ISO14584||
|Washer|ISO7089||
|Washer|ISO7090||
|Washer|ISO7092||
|Washer|ISO7093-1||
|Washer|ISO7094||
|Nut|ISO4032||
|Nut|ISO4033||
|Nut|ISO4035||
|Nut|EN1661||
|Nut|DIN557||
|Nut|DIN562||
|Nut|DIN985||
|ScrewTap|ScrewTap||

luzpaz, 2020-02-04:
not a bad idea

luzpaz, 2020-05-20:
Please post this issue to the forum as well + add the URL when you do.

gbroques, 2020-05-21:
> Please post this issue to the forum as well + add the URL when you do.

Sure, any suggestions as to where on the forum?

luzpaz, 2020-05-21:
@gbroques check out https://github.com/shaise/FreeCAD_FastenersWB#feedback

luzpaz, 2021-03-31:
@gbroques did this get any traction?

gbroques, 2021-03-31:
@luzpaz No, I never bothered with providing feedback on the forum.

It seems like this workbench is more geared towards GUI users, and not meant to be use programmatically as a Python module by other workbenches or Python code.