shaise/FreeCAD_FastenersWB #55: Install ulrich1a screw_maker Macro as Python Package

gbroques, 2020-01-13:
I created an [issue](https://github.com/ulrich1a/screw_maker/issues/4) in @ulrich1a [screw_maker](https://github.com/ulrich1a/screw_maker) repository to publish the package on The Python Package Index (PyPI) so the Fasteners workbench can simply install it rather than maintain a copy-pasted file.

The [FreeCAD Workbench Starter Kit](https://github.com/FreeCAD/Workbench-Starterkit) has some discussion on how to do this, and some example projects using this pattern.

See [Using pip](https://github.com/FreeCAD/Workbench-Starterkit#using-pip-setuptool-or-distutlis) and [uploading your package to pypi](https://github.com/FreeCAD/Workbench-Starterkit#uploading-your-package-to-pypi) sections in the README.

Any thoughts on this proposal?

luzpaz, 2020-05-20:
@gbroques better to open a thread on the forum to discuss this. It will gain more traction than here.

gbroques, 2020-05-21:
@luzpaz Thanks for steering me in the right direction, as I'm still very new to FreeCAD and the community.

I'll do this for future issues I create on FreeCAD GitHub repositories as it seems to be preferred by FreeCAD developers.

For now, I'm focused on [ose-3d-printer-workbench](https://github.com/gbroques/ose-3d-printer-workbench) and not very interested in putting forth the effort to move this or other issues I raised on this repository forward. (although I think they're good ideas still)

luzpaz, 2022-03-27:
@ulrich1a is unresponsive. maybe it's time to just consider this WB a legit fork and start refactoring it because screw_maker macro is enormous and would really benefit from it?

gbroques, 2022-03-29:
> @ulrich1a is unresponsive. maybe it's time to just consider this WB a legit fork and start refactoring it because screw_maker macro is enormous and would really benefit from it?

That makes sense to me.

I'll go ahead and close this issue.

luzpaz, 2022-03-30:
Disagree, should keep this open since this a legit feature request

shaise, 2022-03-30:
the WB version is far from being a copy-pasted version. It was too specific to be used as a general library, so there are lots of changes. Each time the macro was updated I had to manually apply all the changes. Right now it is so different, that this method is no longer possible at all.

luzpaz, 2022-03-30:
I wish @ulrich1a would be inspired to hack on this workbench

shaise, 2022-03-30:
Well, he is marked as a collaborator...

gbroques, 2022-03-31:
> the WB version is far from being a copy-pasted version. It was too specific to be used as a general library, so there are lots of changes. Each time the macro was updated I had to manually apply all the changes. Right now it is so different, that this method is no longer possible at all.

Thanks shaise, that helps clarify things.

@luzpaz @shaise Should we close this issue then?