shaise/FreeCAD_FastenersWB #56: [Feature Request] 5/8 inch 27 threads per inch

burtbick, 2020-02-13:
Would it be possible to add generation for a 5/8 inch diameter with 27 threads per inch?  This is for the US microphone stand thread.

I was looking at trying to do it myself but was hitting some issues with getting the code to build in my configuration.  

I thought I'd seen a comment somewhere that this was something that was being looked at, but haven't been able to find it again.

shaise, 2020-02-16:
The fastener workbench rely on screw_maker macro to generate the bolts. unfortunately screw_maker does not support US standard.

ferdymercury, 2020-07-01:
I would like to upvote this feature request. For example, I need to add a thread M42 x 1.0 mm which is very common in optic lens tubes. It would be nice to be able to specify the pitch in the 'data tab', together with the length.

ferdymercury, 2020-07-01:
Related: https://github.com/shaise/FreeCAD_FastenersWB/issues/52 
https://github.com/shaise/FreeCAD_FastenersWB/issues/26

Maybe, it would be nice to add just a 'custom screw' option, where the user has to input all data, considering it is not in the standard. So that you do not need to modify the standard data tables every time a user comes up with a new weird screw/bolt.

ferdymercury, 2020-07-13:
Alternatively, one could use just the standard equations, so that the thread geometry is calculated on-the-fly:

https://www.fastenerdata.co.uk/media/wysiwyg/screw-threads/fine_pitch_thread.jpg

ferdymercury, 2021-04-01:
See also https://github.com/shaise/FreeCAD_FastenersWB/issues/67#issuecomment-723028691