shaise/FreeCAD_FastenersWB #58: Add LGTM badge

luzpaz, 2020-03-01:
https://lgtm.com/projects/g/shaise/FreeCAD_FastenersWB/?mode=list

```
[![Total alerts](https://img.shields.io/lgtm/alerts/g/shaise/FreeCAD_FastenersWB.svg?logo=lgtm&logoWidth=18)](https://lgtm.com/projects/g/shaise/FreeCAD_FastenersWB/alerts/)
```

Example: [image]

luzpaz, 2020-03-01:
Also
```
[![Language grade: Python](https://img.shields.io/lgtm/grade/python/g/shaise/FreeCAD_FastenersWB.svg?logo=lgtm&logoWidth=18)](https://lgtm.com/projects/g/shaise/FreeCAD_FastenersWB/context:python)
```

Example: [image]

shaise, 2020-03-01:
where to add it to?

luzpaz, 2020-03-01:
I'll send a PR :wink:

PR #59 (2020-03-01): README: Add LGTM badges
Closes #58
commit 0184c05d (2020-03-01): README: Add LGTM badges

Closes #58
closed 2020-03-01 by shaise