shaise/FreeCAD_FastenersWB #60: BOM doesn't count correct

hpbmo, 2020-03-26:
Hi

I like the way, how I can place fasterners. Now I would like to create a BOM for my delta printer. 

When I group washers and screws, ect., BOM wil count attached component twice. Please look inthe forum:

https://forum.freecadweb.org/viewtopic.php?f=3&t=44577

Please, can you confirm this behavior?

Thanks

Hanspeter

shaise, 2020-04-14:
As I said, grouping makes issues. Please attach a sample file.

shaise, 2020-04-14:
I think I fixed it. please check

hpbmo, 2020-04-15:
Great, it counts correct, if the the screws etc. are in groups. Within my 3d printer design I use a lot of parts and groups. There also a lot screws, nuts and washers organised in groups. The fasterners where all placed with an older version expect 3 M4 screws with custom length 50mm. And this 3 screws show now an error:

`FSSCrewObject` object has no attribute `itemText`

The file is big, 13MB. If you are interessted, I will attach it.

With an other smaller design I find an issue with using Part. If I drop all Bodies inside the Part, the fasteners appears doubled in the model tree. If you try this with the attached file, you will see what happens. It's difficult to explain 

[Velotraeger v1.3.zip](https://github.com/shaise/FreeCAD_FastenersWB/files/4479427/Velotraeger.v1.3.zip)

Thanks again for your work

shaise, 2020-04-15:
when do you get the 'itemText' error? when you create a bom?

hpbmo, 2020-04-15:
The screw has an red exclamation mark at the symbol and when point on it there is the message poping up
[image]

shaise, 2020-04-15:
Hi @hpbmo 
I have fixed some issues now that can be related to this bug. You might need to recreate the screw again, since it might be corrupted. please test and tell me
shai.

hpbmo, 2020-04-15:
Hi @shai 

I had really to recreate the screw. Also, I had to recreate the washers wich doesn't base on a 'PartDesign Body' or a on 'Part Body'. If the base Object is a 'PartDesign Feature' it brakes the design.
Are the fasteners placed inside a Part, then quantity is still not correct. The washer are counted twice. They had to be placed inside the part, because it's rotated. Otherway they don't follow the Body.
 
hpb

[Velotraeger v1.3.FCStd.zip](https://github.com/shaise/FreeCAD_FastenersWB/files/4483340/Velotraeger.v1.3.FCStd.zip)

shaise, 2020-04-15:
Hi @hpbmo 
Regarding placing objects under parts, I am sorry, but I am not familiar enough with how this works inside FfeeCAD, so it is hard for me to fix.
shai

shai, 2020-04-16:
> Hi @shai
>
> I had really to recreate the screw. Also, I had to recreate the washers wich doesn't base on a 'PartDesign Body' or a on 'Part Body'. If the base Object is a 'PartDesign Feature' it brakes [...]

Hehe... Wrong person mention 😉

hpbmo, 2020-04-16:
Hi @shaise

> Regarding placing objects under parts, I am sorry, but I am not familiar enough with how this works inside FfeeCAD, so it is hard for me to fix.

You did a very good job with fixing the problem with the group, this helps a lot. And maybe you find a way. 

My programming skills have remained at basic and turbo pascal some years ago ;-). So I have no idea, how I could help.

Thanks again
hpb

luzpaz, 2020-05-20:
I'm confused, the original issue has been fixed ?

hpbmo, 2020-06-29:
Hi @luzpaz 

Sorry for the late response. I was quiet busy the last few weeks and didn't do anything with freecad / fastenersWB.
As far as I remember, the counting works fine if the fasteners are in the root tree or inside a group. As soon as you drag one of them inside a part the counting fails. I try it again in the next few days and post the result

hpb

karliss, 2021-10-21:
Just got one more similar case of overcounting, in my case it was due to subShapeBinder.  Adding one more case checking for `parent.TypeId != 'PartDesign::SubShapeBinder'` seemed to help.

luzpaz, 2022-02-07:
@shaise :point_up:

closed 2022-08-31 by shaise