shaise/FreeCAD_FastenersWB #61: Please increase M3 range to 35mm for ISO4762 screws

glabifrons, 2020-04-12:
Please extend the length range of the ISO4762 screws to 35mm (currently 30mm). I'm modeling around parts from an extruder kit that uses M3 socket-head cap-screws of that length, so I'm sure they exist. :)

I found the line, I can try and make a pull request if you'd like.
https://github.com/shaise/FreeCAD_FastenersWB/blob/master/screw_maker.py#L1082

Thank you very much for creating this extremely useful addon! :)

shaise, 2020-04-12:
Hi,
The fasteners workbench is actually based on the screw_maker macro made by @ulrich1a. He made these fasteners based on the written standard (ISO4762) in this case. I know M3x35 exists and also M3x40. But it is not per the written standard, so I'm not sure if he will add them.
shai

easyw, 2020-04-12:
@shaise It would be nice to have an option to add 'user' model saved locally

shaise, 2020-04-12:
@easyw ,
I thought about that, I think that instead of adding 'user' model which involve table manipulations and savings, just give an option to enter arbitrary fastener length - most cases this is what is missing.

glabifrons, 2020-04-12:
> @easyw ,
> I thought about that, I think that instead of adding 'user' model which involve table manipulations and savings, just give an option to enter arbitrary fastener length - most cases this is what is missing.

I love that idea! I was just trying to make it as easy as possible (1 character change). :)

shaise, 2020-04-13:
I have added the option for custom screw lengths.
Please test and let me know if you see any issues. Specially, test if it does not brake old files, saved with prev version of fastemer WB.
Thanks!

hpbmo, 2020-04-14:
I can't add any washer or nut.

Report view:

`Running the Python command 'FSISO7089' failed:
Traceback (most recent call last):
  File "C:\Users\Hanspeter Blaser\AppData\Roaming\FreeCAD\Mod\fasteners\FastenersCmd.py", line 244, in Activated
    FSScrewObject(a, self.Type, selObj)
  File "C:\Users\Hanspeter Blaser\AppData\Roaming\FreeCAD\Mod\fasteners\FastenersCmd.py", line 53, in __init__
    self.VerifyMissingAttrs(obj, diameters[1])
  File "C:\Users\Hanspeter Blaser\AppData\Roaming\FreeCAD\Mod\fasteners\FastenersCmd.py", line 62, in VerifyMissingAttrs
    slens = screwMaker.GetAllLengths(obj.type, diameter)
  File "C:\Users\Hanspeter Blaser\AppData\Roaming\FreeCAD\Mod\fasteners\ScrewMaker.py", line 207, in GetAllLengths
    range = screwTables[type][3][diam]

'NoneType' object is not subscriptable`

Tested with
OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.20514 (Git)
Build type: Release
Branch: master
Hash: a05ffd499366ab12fa1b43f37db73f4f1e9c332d
Python version: 3.8.2
Qt version: 5.12.5
Coin version: 4.0.0
OCC version: 7.4.0
Locale: German/Switzerland (de_CH)

I like your addon and use it quite often

shaise, 2020-04-14:
I knew i will miss something.... Should be fixed now, please update.
shai

hpbmo, 2020-04-14:
Wow, you're realy fast

Nut and washer are ok now, only screw tap has still an error

`Running the Python command 'FSScrewTap' failed:
Traceback (most recent call last):
  File "C:\Users\Hanspeter Blaser\AppData\Roaming\FreeCAD\Mod\fasteners\FastenersCmd.py", line 462, in Activated
    FSScrewRodObject(a, selObj)
  File "C:\Users\Hanspeter Blaser\AppData\Roaming\FreeCAD\Mod\fasteners\FastenersCmd.py", line 390, in __init__
    self.VerifyCreateMatchOuter(obj)

'FSScrewRodObject' object has no attribute 'VerifyCreateMatchOuter'`

shaise, 2020-04-14:
oops... sorry about that. Please try now.
I'm trying to respond fast since this is a new feature and I want to make sure I didn't break things that worked before. (which unfortunately I did, as you can see ;)

hpbmo, 2020-04-14:
The screw tap is now also ok. But ..... I forget until now to check the BOM generation. I addead one of each element and tried to get a BOM. And it fails.

`Running the Python command 'FSMakeBOM' failed:
Traceback (most recent call last):
  File "C:\Users\Hanspeter Blaser\AppData\Roaming\FreeCAD\Mod\fasteners\FastenerBase.py", line 674, in Activated
    method(obj, cnt)
  File "C:\Users\Hanspeter Blaser\AppData\Roaming\FreeCAD\Mod\fasteners\FastenerBase.py", line 713, in AddStud
    self.AddFastener("PEM Stud " + obj.diameter + "x" + obj.length)

AddFastener() missing 1 required positional argument: 'cnt'`

When you investigates on this, maybe you can have a look at issue #60? But this is not really urgent ;-). I think this needs much more work.

Thanks again

shaise, 2020-04-14:
Bom + grouping is a problem, I know. see issue #50.
regarding your crash, do you have studs in your model?

hpbmo, 2020-04-14:
I don't have studs in my model. Maybe this little test file will help you
[Test_Fasterners.zip](https://github.com/shaise/FreeCAD_FastenersWB/files/4474504/Test_Fasterners.zip)

shaise, 2020-04-14:
and this file crashed when creating a bom?
I know there was a poroblem (i just fixed) with bom and studs. The error you sent clearly state the problem is with "AddStud".
Anyway is this an example of the group problems?

hpbmo, 2020-04-14:
This file has crashed, but now, creating BOM starts fine. But you can use this file for the group problem. If you need a file with more screws, nuts, etc. I could add another one. Should I do this in the other issue?

shaise, 2020-04-14:
should work now for groups. please test

luzpaz, 2023-03-06:
@hpbmo ping  
Also is there anything left to do in this ticket ?

luzpaz, 2023-03-10:
Closing this ticket. If incorrect, please re-open

closed 2023-03-10 by luzpaz