shaise/FreeCAD_FastenersWB #62: [enhancement] standoff+spacers option

easyw, 2020-04-14:
For electronics assemblies it would be nice to add [standoff & spacers](https://www.we-online.com/catalog/en/SSTIE_STEEL_SPACER_STUD_METRIC_INTERNAL_EXTERNAL/?sq=standoff+spacers&sp=https%3A%2F%2Fwww.we-online.com%2Fweb%2Fen%2Fwuerth_elektronik%2Fsearchpage.php%3Fsearch%3Dstandoff%2Bspacers) like in the linked documentation
[image]

shaise, 2020-04-15:
Good idea. Hope to have time for it soon. A bit busy now with my [LinuxCNC](https://github.com/shaise/GrblCNC) clone now.

easyw, 2020-04-15:
@shaise 
thanks for having a look. 
Spacers and Standoff are available as [CadQuery python scripts](https://github.com/easyw/kicad-3d-models-in-freecad/tree/master/cadquery/FCAD_script_generator/Standoff_FF_MF), but having a python feature in FC as for the screws would be the best.

shaise, 2020-04-15:
Done.

easyw, 2020-04-15:
@shaise 
that was fast! :smile:
Would it be possible to also add spacers (which are a sub case of standoffs)?
[spacers](https://www.we-online.com/catalog/en/SSTII_STEEL_SPACER_STUD_METRIC_INTERNAL_INTERNAL/)
[image]

shaise, 2020-04-16:
Done as well. Only because you are the one who asked....

easyw, 2020-04-16:
Thanks a lot!!! Great Tool 😸 
I 've found a minor issue (spacer without icon) at this line:
https://github.com/shaise/FreeCAD_FastenersWB/blob/69e5e04e268642e7cc177b746b643469980a5278/PEMInserts.py#L997
should be changed with:
`FastenerBase.FSClassIcons[FSPcbSpacerObject] = 'PCBSpacer.svg'
`
thanks again, this will be very useful for KiCAD integration 😄

shaise, 2020-04-16:
ooops... fixed.

easyw, 2020-04-16:
thx again 😺

closed 2020-04-16 by easyw