shaise/FreeCAD_FastenersWB #63: Error Step Export with use of Fasteners Workbench

mikereiner, 2020-05-14:
I want to export my assembly module, but there occures an error. After some checks, i think it depends on the connection of the screw and the hole. 
When i insert the screw directly to a hole. After that, i move the screw with drag & drop to my assembly-group (part). Now, i want to export as step-file, but an error occurs.
When i insert the screw without connection and move with drag & drop to the part. I connected it with the a2+ workbench. Here there is no error while exporting as step-format. 

Does anybody have a similar Problem, or can handle this problem? 
Thank you!

shaise, 2020-05-14:
I have no idea how freeCad  handles connected parts internally. I think you should post your question in the forum: https://forum.freecadweb.org , there you can upload a sample file with the error.

mikereiner, 2020-05-14:
okay, thank you for your help!

luzpaz, 2022-02-02:
@mikereiner did you ask? Can you x-post to here the URL of the forum thread?

luzpaz, 2022-05-03:
@danmiel since this A2plus related, would you mind weighing in?

DanMiel, 2022-05-03:
I could try looking at the code but I have not had coding education and it takes me a while to understand Kbwbe’s code. 
I almost didn’t use A2plus because of the time it takes to create a constraint. Time is not just the time it takes to select the features. In A2plus time = click the first feature + select control key + select second feature + mouse travel to selection icons + select icon + travel to  “Accept” button + select “Accept” button. Add to that, time I have to wait until my tremors settle down to select. After more than 20 years of using 3D software I’m used to selecting two features and the constraint is created.
I know that sounds like an efficiency person but that is the way engineers look at 3D software, to many mouse clicks and too much mouse travel = software doesn’t get used.
Other items that was hard to accept, is the lack of diagnostic tools for constraints, time spent rotating the model to select features on the backside and the inability to select a feature in the model.
Things I do like: building part models outside of the assembly program and being able to create a bill of materials. 
With all of that said, I would like to help with questions concerning A2plus and its use but I don’t understand a lot of the coding and none of the mathematics he uses so I seldom change his code, which is why my code is options and not part of his code.
I hope this explains my willingness to help and my limitations.
Dan Miel

luzpaz, 2022-05-04:
@mikereiner ping

luzpaz, 2022-05-31:
@mikereiner bump

closed 2023-03-06 by luzpaz