shaise/FreeCAD_FastenersWB #64: [Feature Request]  Die to Make external Thread

cram79, 2020-05-15:
Would it be possible to add a 'die' tool/feature that allow the making of a external thread on a existing shaft, also with a selection of thread length?

shaise, 2020-05-15:
Interesting idea. I'm rather busy now, but I will add to the "todo" list.

slowinternets, 2020-05-20:
OK I gave it a shot:
I added a 'thread-die' object that implements what I think you are looking for. It starts with a hollow pipe and subtracts a length of outer diameter thread, to make an OD thread-cutting tool as requested. When modeled-thread is False, the inner diameter of the pipe becomes the major diameter of its thread size. 

While I was at it, I also added a threaded rod object. If are using some standard threaded rod in a project, you can skip modelling a shaft and using the die object on it.

I was pleasantly surprised to see that the screw maker macro actually models a specifically inner diameter style thread when the tap tool is used. I left this functionality unchanged. Both the die and threaded rod objects use the proper OD threads.

[image]

Some new and updated icons:

[image]

My fork with the changes:
61a108d37011435b0571be5014a259aef1ab69d1

Other things to note:
-The Die and Threaded Rod objects are not currently added when generating a BOM table.
-Warnings abut the memory/CPU intensive nature of modeled threads still apply. A 1m length of M6 threaded rod took 3 GB of ram and several minutes to compute on my PC.
-I tested the new features on V0.18 and 19, but I'm sure I missed a few bugs.

Check it out and tell me what you think!
Thanks, 
Alex

cram79, 2020-05-21:
Hi Alex, what you describe is what I meant, that looks great, thank you for taking the time.  How can I use what you have done though?  ie I take it at the moment I can't just update the WB through Add-on Manager within FreeCAD.

slowinternets, 2020-05-21:
I have added a pull request for the changes. If/when it gets accepted, you will be able to use the new features through the addon manager. 

If you are feeling impatient, you can download https://github.com/slowinternets/FreeCAD_FastenersWB as a zip file, and copy its contents into the correct directory. 
On windows, it should be C:\user\Appdata\Roaming\FreeCAD\Mod\fasteners
Linux: ~/.FreeCAD/Mod/festeners
Extract  the zip file and copy all of its contents to the proper directory.
For a more thorough tutorial, see: https://wiki.freecadweb.org/How_to_install_additional_workbenches

shaise, 2020-05-21:
@slowinternets ,
looks great! i'm on vacation now, I will take a look as soon as i return

cram79, 2020-05-22:
@slowinternets ,    Thank you again.

cram79, 2020-05-22:
@slowinternets,   Thought I try and install and give it a spin, its great. But I just realised though when I went to use it that the icon and description don't match the function. ie The 'ScrewTap' tool icon creates a external thread, where a Tap should create a inner thread.       I'm not sure the Tap function is working as expected (with ignoring the icons being dis-orientated) . For instance it seems to just create a rod with a inner thread, not actually cut a inner thread on a solid cylinder for instance.      Sorry I'm greatly appreciative, just trying to add input.

slowinternets, 2020-05-22:
I am fairly certain that it works as intended, but I recognize that the logic behind these tools may be a bit unintuitive. I will try to explain more thoroughly:

The best way to think about it is that the screwTap and threadDie tools literally create the objects they are named for. Adding a screwTap object creates an actual thread cutting tap. But because we are working in CAD software and not real life, it doesn't quite look like a real tap. The same explanation applies for the threadDie.
[image]
[image]
Here is my workflow for using these tools in FreeCAD:

Start by creating some objects. A tap, a die, and a shaft/block for us to cut threads on.
[image]

Now, us the part workbench csg tool to subtract the die from the shaft, and the tap from the block:
[image]

Translate the new objects for a visual check that they fit together nicely.
[image]

Let's zoom in for a closer look:
[image]

We can see that the screwTap has created clearance at the top/bottom of the threads so that the threaded rod fits better. This is because the screwTap was generated with a different profile, specific to Inside Diameter threads.

the workflow for creating complex objects with CSG operations can certainly seem unintuitive. I hope this explanation was helpful.

Addendum: 
when creating a shaft for use with the ThreadDie, I recommend making its diameter slightly larger or smaller than the diameter of the thread you want to cut (IE: 6.1mm or 5.9mm for an M6 thread). The openCASCADE software that actually calculates the subtraction operation tends to behave poorly when you try to work with exactly matching diameters.

cram79, 2020-05-23:
@ slowinternets ,   Thank you for the explanation I now understand that it is actually creating the 'tool' as such and than need to make a boolean cut to get the desired result. Yes it didn't seem intuitive to me to work that way. I was expecting to select circular edge of a cylinder then select the Die tool and have it 'cut the thread' resulting in a external thread.

slowinternets, 2020-05-25:
The new tools have been merged into the main branch, and are now available through the addon manager. 

It would certainly be very cool to be able to go directly from a selected edge or face to a threaded part. However, that would be a much larger and more complicated feature to implement. For now, there are other things that I would like to add to the workbench first. If you are happy with the state of these tools at the moment, please consider closing this issue.

ferdymercury, 2020-06-30:
Very nice tool, thanks so much! I have worked with it and have found a bug when working with more complex examples. I do not know if the bug is from FreeCAD directly, or something in this workbench? See https://forum.freecadweb.org/viewtopic.php?f=3&t=48241&p=412994

ferdymercury, 2020-07-01:
Ok apparently, it was due to a 'coplanar issue'.

luzpaz, 2021-03-31:
What's left regarding the FR ?

cfunseth, 2021-08-11:
@alexneufeld Wonderful addition to the workbench, thank you!

Would it be difficult to add imperial sizes for the tap and die tools?  I'm not familiar with how plugins are written but I'll jump in if someone here isn't able to figure it out easily.

alexneufeld, 2021-08-12:
Yeah, I could do that. Busy at the moment, give me a few days to throw a PR together

luzpaz, 2022-02-02:
Can we close this ticket?

berberic2, 2022-08-23:
> Can we close this ticket?
I would say: yes.

shaise, 2022-08-23:
Closing this issue as complete

alexneufeld, 2022-12-27:
@shaise reminder to close this issue

closed 2022-12-27 by shaise