shaise/FreeCAD_FastenersWB #67: Add comments to the size tables

Ivan-Monahhov, 2020-07-02:
I am using FreeCad to design a few things for my hobby.  I need some weird threads mostly M9x0.5

I thougth to add these threads myself , but I cannot figure out what which size does in the tables.

ulrich1a, 2020-07-05:
It is P for pitch of the thread. But be aware, that  there is no warranty, that you get a valid solid. The thread creation with the OCCT-cad kernel is a fragile operation.

Ulrich

ferdymercury, 2020-07-20:
I created an excel spreadsheet that might help you in this regard to identify what is what. You just need to Specify 9 as diameter and 0.5 as pitch. The rest is auto-calculated.

[ThreadCalculator.ods.zip](https://github.com/shaise/FreeCAD_FastenersWB/files/4947264/ThreadCalculator.ods.zip)

[image]

luzpaz, 2020-10-18:
@Ivan-Monahhov any progress on this?

ferdymercury, 2020-11-05:
In the meanwhile, I created a FreeCAD file that allows you to specify in a spreadsheet a custom diameter, pitch and number of turns and it creates a female thread, using an Helix and a sketch / Frente additive pipes.

To change it to a male instead of female thread, you might need to swap the sketch lines that are construction and normal lines.

[image]

[ThreadCalculator.FCStd.zip](https://github.com/shaise/FreeCAD_FastenersWB/files/5495411/ThreadCalculator.FCStd.zip)

ferdymercury, 2020-11-06:
Fixed version here. [ThreadCalculator.FCStd.zip](https://github.com/shaise/FreeCAD_FastenersWB/files/5500757/ThreadCalculator.FCStd.zip)

Thread, pitch and height are specified in the Spreadsheet. The rest is automatic.

[image]

[image]

ferdymercury, 2020-11-06:
Here the equivalent FC file for custom male threads

[ThreadCalculator_Male.FCStd.zip](https://github.com/shaise/FreeCAD_FastenersWB/files/5500962/ThreadCalculator_Male.FCStd.zip)

[image]

muenznej, 2021-04-11:
that was pretty helpful, thank you. I hope it will be merged/added soon, its definitely not as exotic for hobbiest than one would think.