shaise/FreeCAD_FastenersWB #71: Problem with M30 DIN 975 threaded rod

TedMarshall, 2020-09-01:
I created an M30 x 20mm rod. Without threads, it is the proper length. However, when the threads are added, the bottom 2mm are missing. Continuing to test longer lengths, 21mm is correct but 22, 23 and 24mm all truncate at 21mm. 25, 26 and 27mm truncates at 24.5mm (approximate). 28mm is correct. That is the limit of my testing.

I did also try an M30 ISO 4762 Hex Socket Head Cap Screw and it did not exhibit any odd behavior, at least at those lengths.
[Fasteners.zip](https://github.com/shaise/FreeCAD_FastenersWB/files/5154038/Fasteners.zip)

shaise, 2020-09-01:
the fasteners workbench is a wrapper over the screwmaker macro, and the thread code is taken from there. I believe the reason for truncation is that because the threaded rod is using an integer multiple of the thread pitch which is 3.5mm hence 21, 24.5 and 28 works correctly, and others truncate. As a workaround you can create a longer thread and using boolean to cut the excess length.

alexneufeld, 2022-12-27:
This should be closed. As of #258, the truncation behavior is no longer a property of anything in the codebase.

shaise, 2022-12-27:
@TedMarshall , can you please verify and close?