shaise/FreeCAD_FastenersWB #72: PCB Metric Standoff

leoheck, 2020-09-02:
Hi, I am trying to use the PCB Metric Standoff, but it does not allow me to change settings freely. Only few options on the list.
For instance, I would like to create this one, but the lengths available are 10 mm and 12 mm. The length is a list and not an input field. There is no way to type 11 mm there.

[image]

shaise, 2020-09-03:
Hi @leoheck ,

I have just added a feature that lets you set custom lengths. Please update, try, and let me know. 
Note: If you are loading an old file with standoffs of previous version, you need to recompute them for this setting to appear.

shai

leoheck, 2020-09-03:
Yay! This is nice. I will try it right now @shaise

leoheck, 2020-09-03:
It is working just fine but I have another and related question.
The threaded part of this standoff, is 6mm, how do I set that?

I am thinking it is on width. But I can't change the `5` there.
[image]

shaise, 2020-09-03:
Sorry, you can't. This is the standard. the width is not related. the width is for M3 standoff. There are several width standards (the hex body thickness)

easyw, 2020-09-10:
@shaise I think @leoheck was referring to the length of the screw part of the standoff... some manufacturer have a slightly different length i.e. instead of 6mm, 4mm or 5mm

leoheck, 2020-09-11:
Yes, that's right @easyw 
Do you think it is possible to add a custom length for that treaded part as well?

shaise, 2020-09-11:
Done.

closed 2020-09-21 by leoheck