shaise/FreeCAD_FastenersWB #73: Error: unknown dialect

lucaivan, 2020-09-03:
Hello,

I have just installed the fasteners workbench through the addon manager but when I try to use it gives me an error saying unknown dialect. In report view the following appears:
`unknown dialect`
`Traceback (most recent call last):`
  `File "<string>", line 46, in Initialize`
 ` File "/home/luca-ivan/.FreeCAD/Mod/fasteners/FastenerBase.py", line 30, in <module>`
    `from screw_maker import *`
  `File "/home/luca-ivan/.FreeCAD/Mod/fasteners/screw_maker.py", line 136, in <module>`
    `itemdict = csv2dict(itempath,fieldsnamed=True)`
  `File "/home/luca-ivan/.FreeCAD/Mod/fasteners/screw_maker.py", line 116, in csv2dict`
    `reader = csv.reader(data, skipinitialspace=True, dialect='unix', quoting=csv.QUOTE_NONNUMERIC)`

It seems that there is an error with csv but I don't know how to solve it. I use Linux Debian and FreeCAD 0.18
Thank you for your time.

shaise, 2020-09-03:
I have no linux to test it, can you try it on freecad 0.19?

lucaivan, 2020-09-04:
Thanks very much, in freecad 0.19 works perfectly

closed 2020-09-04 by lucaivan

Asgaurd36899863, 2021-01-15:
Hey! from where & how did you install freecad 0.19?