shaise/FreeCAD_FastenersWB #74: Self Tapping Screws

leoheck, 2020-09-16:
Hello, 

I was trying to find these self-tapping screws that work really well with plastics.
For instance, size M2.6 that I need to use with a pointy tip. 

[image]

My reference is here.
https://www.amazon.com/Tapping-400Pcs-Screws-Assortment-Plastic/dp/B07VQZXGH2

Can you tell me if we have this in this workbench?
Also, if it is not there yet, is it possible to ask this as a feature request?

leoheck, 2020-09-21:
@shaise do you know anything about these self-tapping screws or at least if it is possible to use M2.6 somewhere?

shaise, 2020-09-21:
Sorry, but no self tapping screws yet. The fasteners WB is based on the screw-maker macro, and it does not have it either.

leoheck, 2020-09-21:
Thanks for letting me know. Where I can find this screw-maker macro. Is it possible to ask this as a feature request there? Or even better, offer any kind of help there?

shaise, 2020-09-22:
https://github.com/ulrich1a/screw_maker. not sure how active it is.

leoheck, 2020-09-22:
The project has a couple of years but the latest update was 3 months ago. It looks that is not forgotten. I am going to make a request there. Lets see.

FreeCAD-Tools, 2022-09-27:
It seems to me as a basis for Self Tapping Screws may can use DIN571 screw. Just need to change the cap.
But I do not competence in the code of the formation of screws.

[image]

luzpaz, 2022-11-14:
Perhaps related #138

shaise, 2022-11-14:
DIN571 is implemented now.

EA32, 2022-11-14:
[image]

At this moment, GOST 1144(2) and GOST 1144(4) Wood Screws is available, they are not is Self-Tapping Screws, but have thread with very similar profile and along the entire length. Their have semi-sphere heads.

luzpaz, 2023-03-06:
@alexneufeld any possibility you'd be interested in implementing this ?

alexneufeld, 2023-03-06:
Sure

hasecilu, 2023-12-23:
Hi, I sent a PR for some self taping screws if someone want to check/test.

closed 2024-01-12 by shaise