shaise/FreeCAD_FastenersWB #75: M10 screws with "fine" pitch instead of "coarse"?

johnliu55tw, 2020-10-09:
Hi, can I switch the pitch of M10 screws to "fine" (1.0mm)? Seems like the default is "coarse" (1.5mm).

ferdymercury, 2020-11-05:
Not yet, though there are several feature requests to have more flexible selection of threads. https://github.com/shaise/FreeCAD_FastenersWB/issues/67

EvilGremlin, 2021-06-15:
Second that, custom thread pitch is essential feature that should've been there from the very beginning... I love Fasteners, but absence of custom pitch is killing me every time :(

berberic2, 2023-09-12:
Has this been solved with #285?
see. #272.

EvilGremlin, 2023-09-12:
Complely forgot about that... yeah, works for a long time, though only for rod etc
Would be nice to have completely customizable screw and nut too, with custom head diameter and socket size.
[image]