shaise/FreeCAD_FastenersWB #77: Error in FreeCAD 0.18~pre1+dfsg1-5 (Debian stable)

morfeus2, 2020-10-17:
First of all thanks for the great job: it would be impossible to work with a CAD that misses some kind of fasteners generator utility.
I just tried to upgrade from an old versione of the WB, since now I need PCB spacers and standoffs: the present master version fails to initialize, when I first select the WB.

I am no expert in git, but I tried to fine more details, by getting different versions based on commits (I found no other way of getting old versions since I think you did not published any tags of branches); if I got correctly the updated V0.3.30 (the first showing a subdir "FsData") is the first one failing with:

```
FreeCAD 0.18, Libs: 0.18R
© Juergen Riegel, Werner Mayer, Yorik van Havre 2001-2019
  #####                 ####  ###   ####
  #                    #      # #   #   #
  #     ##  #### ####  #     #   #  #   #
  ####  # # #  # #  #  #     #####  #   #
  #     #   #### ####  #    #     # #   #
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

unknown dialect
Traceback (most recent call last):
  File "<string>", line 46, in Initialize
  File ".FreeCAD/Mod/FreeCAD_FastenersWB-V0.3.30/FastenerBase.py", line 30, in <module>
    from screw_maker import *
  File ".FreeCAD/Mod/FreeCAD_FastenersWB-V0.3.30/screw_maker.py", line 136, in <module>
    itemdict = csv2dict(itempath,fieldsnamed=True)
  File ".FreeCAD/Mod/FreeCAD_FastenersWB-V0.3.30/screw_maker.py", line 116, in csv2dict
    reader = csv.reader(data, skipinitialspace=True, dialect='unix', quoting=csv.QUOTE_NONNUMERIC)

```

V0.3.22 seems to run fine. The only problem in this version is that it has fixed lengths and the list of available possibilities is quite limited (I guess it matches the available models from a producer, but other brands produce also different lengths).

I tried to patch locally, by starting from V0.3.22 and then copying the updated PEMInserts.py from here:
https://github.com/shaise/FreeCAD_FastenersWB/blob/811d3cc1225b12342c165e3c70e3a35c1da6ee15/PEMInserts.py

And seems to work properly for PCB standoffs, so, since I am so close to the goal, if you could further modify PEMInserts.py to have the possibility to use also custom length also for PCB spacers it would be great: I could copy to V0.3.22 and be happy.

shaise, 2020-10-17:
for the error message please try freecad 0.19 latest build.
The latest version of fasteners do have custom lengths of pcb standoffs (but not pem standoffs)

luzpaz, 2021-04-03:
@morfeus2 ping

luzpaz, 2021-07-02:
recommend closing ticket

shaise, 2021-07-04:
closed due to inactivity

closed 2021-07-04 by shaise