shaise/FreeCAD_FastenersWB #78: Can’t generate ISO 14583 M4×10

berberic2, 2020-10-18:
M4 × 10 does not work. 
M3 × 10, M5 × 10, M4 × 8 and M4 × 12 work

Traceback (most recent call last):
  File "…/.FreeCAD/Mod/fasteners/FastenersCmd.py", line 160, in execute
    s = screwMaker.createFastener(fp.type, d, l, threadType, True)
  File "…/.FreeCAD/Mod/fasteners/ScrewMaker.py", line 337, in createFastener
    return self.createScrew(type, diam, len, threadType, shapeOnly)
  File "…/.FreeCAD/Mod/fasteners/screw_maker.py", line 954, in createScrew
    screw = self.makeIso7045(ST_text, ND_text,l)
  File "…/.FreeCAD/Mod/fasteners/screw_maker.py", line 1474, in makeIso7045
    headShell = aWire.revolve(Base.Vector(0.0,0.0,0.0),Base.Vector(0.0,0.0,1.0),360)
<class 'Part.OCCError'>: BRepAdaptor_Curve::No geometry

shaise, 2020-10-18:
You are right! should be fixed now.

berberic2, 2020-10-18:
Works now.
Thanks

closed 2021-03-31 by berberic2