shaise/FreeCAD_FastenersWB #82: When I try to open the fastenersWB I get a error [ workbench faliure: unknown dialect ]

Asgaurd36899863, 2021-01-15:
Why is it so,
Please check & reply.

Asgaurd36899863, 2021-01-16:
And yes, 
I saw the same issue in the closed issues, 
In that your reply was to try it in freecad 0.19, 
so from where and how to download this version of freecad,

shaise, 2021-01-18:
https://github.com/FreeCAD/FreeCAD/releases/tag/0.19_pre

Asgaurd36899863, 2021-01-19:
In the following link, 
https://github.com/FreeCAD/FreeCAD/releases/tag/0.19_pre, 
The following para, 
_**Windows
Binary developer builds (no compiling necessary) are available. Download the highest revision of the .7z

The builds are portable and do not require an installation. ### Extract the file using 7zip and start the application by double clicking on FreeCAD.exe in the bin sub-directory of the extracted package.**_, 
Where to extract the file

shaise, 2021-01-19:
you can extract it anywhere.
just create a folder under "Documents" called Freecad and extract it there. then go to the bin folder and double click on freecad.exe

luzpaz, 2021-03-31:
@Asgaurd36899863 ping

liyoucaili, 2022-01-18:
我在0.17版本freecad中也遇到这个问题，如何解决

closed 2022-01-19 by Asgaurd36899863