shaise/FreeCAD_FastenersWB #83: Threaded nut issue

GitHub-NuB, 2021-01-19:
I hope this is the right place to post this question, it is my first GitHub post...

I am trying render some nuts for 3D printing. I just upgraded to FreeCAD 0.19 and the latest FastenersWB. I noticed that when I render the threads there are now artifacts in the model.  I just figured out how to manually install workbenches and download/apply old versions of the workbench through GitHub. It appears the issue first appears in the Aug 4, 2020 version.  I included an image of the issue. Is this a known issue? Is it something I'm doing wrong? Thank you in advance!

[image]

shaise, 2021-01-20:
Hi,

The best place for such questions is the freecad forum: https://forum.freecadweb.org 
Please write all the info you wrote here and attach the freecad file that demonstrate the issue.

shai

GitHub-NuB, 2021-01-20:
Thank you for the response. I will try my luck there. Thank you for your efforts on this workbench development!

GitHub-NuB, 2021-01-21:
FYI, just wanted to follow up on this in case anyone else experiences this bug. After a good deal of trial and error, I think I narrowed down the issue. It appears to be related to the 'Nut Scaling' option in the Fasteners toolbench preferences. I was testing with a M10 nut. If I recompute using a 'Scale B' value of 0.360, there is no artifact (and the part passes the geometry check) , if I change the variable to 0.370 the artifact reappears (I originally was trying 0.400 since I was having some issues getting my prints to mate) , in addition the part no longer has valid geometry using the Part 'check geometry' feature.

shaise, 2021-01-21:
Hi, 

Working with scales that high can indeed cause issues. The underlying geometry engine is not very stable when it comes to helix geometry. The b scale should not be that high. Try getting your initial scale with scale A (a multiplier) then use a smaller scale b.

shai

GitHub-NuB, 2021-01-21:
I appreciate the feedback. Now that I know what is causing the issue I can definitely start to work around it.   Regards.

closed 2021-01-21 by GitHub-NuB