shaise/FreeCAD_FastenersWB #86: Feature Request:  Add support for PG threads

sslupsky, 2021-02-09:
It might be useful to add support for PG (Panzergewinde) threads.

sslupsky, 2021-02-09:
So I saw PR #70 that introduced the csv files.  Is this request as simple as creating the spreadsheets?  Is any additional code required to support new threads?

shaise, 2021-02-09:
Threads are extremely difficult without issues. In fact the threads are based on the original macro written by Ulrich, and no one touched it since.

sslupsky, 2021-02-10:
Doesn't Ulrich's macro have parameters to specify the attributes for an arbitrary thread?  Looking at PR #70, it appears that the csv's provide the parameters to render a thread according to Ulrich's macro?  Or, am I mistaken about that?

shaise, 2021-02-10:
The CSV is only done to separate the fasteners parameters from the main screw_maker code. It supports both metric and imperial threads since the cross-section of the thread is the same, just the dimensions are a bit different. If the  PG threads have same cross section, then it might be used for them as well, otherwise a special code will be needed to generate them.

sslupsky, 2021-02-10:
Hmmm, I see.  Is there any documentation regarding what the parameters are?  I see:
```
"Dia","b","P","c","dw","e","k","r","s"
```
I have determined:
"Dia" is the "size" of the thread.
"P" is the pitch

What do the other parameters refer to?

I also found a list of different parameters such as:
```
"Dia","P","t","dp","dt","df","s","z"
```

and
```
"Dia","P","t","dp","df","z","s"
```

sslupsky, 2021-02-10:
PG threads have a different thread angle (80 degrees) versus UTS or Metric (which have a 60 degree thread angle).

shaise, 2021-02-10:
each fastener has its own parameters, based on its shape. some parameters have same meanings some don't.

sslupsky, 2021-02-10:
Ok, thanks for the info.  I think I am seeing the picture now.  The macro is intended to be a very specific tool to generate a very specific type of thread.  Over time, specific thread types were added for specific needs.  Kind of a brute force approach.

I think you are quite correct in your assessment, it would be very tricky to do anything with Ulrich's macro.  Probably needs to be completely refactored.

Closing.

closed 2021-02-10 by sslupsky