shaise/FreeCAD_FastenersWB #88: Additional feature request: extend range - add small metric thread and add ME (Model Engineering imperial) threads

bambukouk, 2021-03-08:
1) I would like to add metric threads smaller than M1.6 currently available in the workbench in model making, watchmaking, small instrument etc (non-standard/smaller hexagon heads, etc)
Examples of fasteners in use:
https://ghw-modellbau.de/index.php?cPath=4_5_7_102&language=de
or even smaller:
https://ghw-modellbau.de/index.php?cPath=4_5_7_103

2)I would like to add  "threaded rod for tapping holes" and "threaded tube for cutting external threads" in a range of ME threads:
https://www.gewinde-normen.de/en/me-thread.html
It's only rod and tube, for use in modelling (there are no "standard" fasteners in this range)

I am more than happy to assist in preparing and entering the data, but my Python skills are "zero" at the moment, hence the request, please.

luzpaz, 2021-06-23:
Hi @bambukouk do you mind x-post to the forum please ? TIA!

bambukouk, 2021-06-23:
> Hi @bambukouk do you mind x-post to the forum please ? TIA!

already there:
https://forum.freecadweb.org/viewtopic.php?f=3&t=56106
and
https://forum.freecadweb.org/viewtopic.php?f=8&t=11429&start=10#p501174

I have managed some additions, but run up against my lack of understanding of the workbench to make more additions,
hence my support for #95

ferdymercury, 2021-08-12:
See also https://github.com/shaise/FreeCAD_FastenersWB/issues/67, including a FreeCAD file where one can create threads of custom diameter / pitches, just by editing spreadsheet parameters.