shaise/FreeCAD_FastenersWB #89: DIN EN ISO 10642 added M2 and M2.5

esopalumpa, 2021-03-15:
If this site is right, M2 and M2.5 has been added to ISO 10642:
```
Änderungen:
Schrauben aus Edelstahl hinzugefügt M 2 und M 2,5 hinzugefügt
Bei M14 bis M20 die Gewindelänge bei Schrauben mit Schaft auf 3d erhöht
```
https://shop.dresselhaus.de/de/62-Normen%C3%BCbersicht/inh_id,272,DIN-7991-%3E-DIN-EN-ISO-10642.html

Disclaimer: I have no access to the norm.

Gspohu, 2021-04-28:
I can confirm there has been added in the ISO 10642:2019.
[https://www.iso.org/standard/72578.html](https://www.iso.org/standard/72578.html)

diniciacci, 2021-05-17:
I have access to the norm, if someone writes somewhere a small guide on how to add sizes to iso10642def.csv i can do a pull request for it. I had a look but the numbers are unclear to me (also because I am mostly a coder).

luzpaz, 2021-06-23:
@diniciacci  You can follow the changes that were made in  https://github.com/shaise/FreeCAD_FastenersWB/pull/91/files

shaise, 2021-11-13:
Added by @alexneufeld.

luzpaz, 2022-03-27:
Closed by 5b7e17d (thanks @alexneufeld!)

closed 2022-03-27 by luzpaz