shaise/FreeCAD_FastenersWB #90: FastenersWB on linked part - External object not supported

nadarbreicq, 2021-03-19:
Hi,

When I try to use FastenersWB on the edge of a linked part, I get an error : `PropertyLinkSub does not support external object`

The complete error on the report is the following : 

```
11:17:14  <Exception> PropertyLinkSub does not support external object
11:17:14  Running the Python command 'FSISO7380-1' failed:
Traceback (most recent call last):
  File "C:\Users\adrie\AppData\Roaming\FreeCAD\Mod\fasteners\FastenersCmd.py", line 254, in Activated
    FSScrewObject(a, self.Type, selObj)
  File "C:\Users\adrie\AppData\Roaming\FreeCAD\Mod\fasteners\FastenersCmd.py", line 39, in __init__
    FSBaseObject.__init__(self, obj, attachTo)
  File "C:\Users\adrie\AppData\Roaming\FreeCAD\Mod\fasteners\FastenerBase.py", line 40, in __init__
    obj.addProperty("App::PropertyLinkSub", "baseObject", "Parameters", "Base object").baseObject = attachTo

PropertyLinkSub does not support external object
```

Does fastenerWB support link part ? Thank you in advance.

```
OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24267 (Git)
Build type: Release
Branch: master
Hash: b2ca86d8d72b636011a73394bf9bcdedb3b109b7
Python version: 3.8.6+
Qt version: 5.15.1
Coin version: 4.0.1
OCC version: 7.5.0
Locale: French/France (fr_FR)
```

shaise, 2021-03-20:
Never tried on linked objects, so it indeed might not support it

boomanaiden154, 2021-03-24:
I can confirm that this is an issue with external linked parts, assuming that I have everything set up correctly. This is a fairly big part of working with assemblies in Assembly4, so if a fix is possible, it would be quite helpful to many trying to use fasteners in multi-file Assembly4 Models. Is there any possible fix/possible plan?

boomanaiden154, 2021-03-25:
A pretty simple fix seems to just be to change over from using `PropertyLinkSub` to `PropertyXLinkSub` on line 40 of `FastenerBase.py`, and this solves issues with external files, but it doesn't position the parts properly, at least in Assembly 4, due to the local coordinate systems that are used for the positioning of parts.

cfunseth, 2021-08-06:
> A pretty simple fix seems to just be to change over from using `PropertyLinkSub` to `PropertyXLinkSub` on line 40 of `FastenerBase.py`, and this solves issues with external files, but it doesn't [...]

I just made this change while using FreeCAD v0.19 and it also works while properly positioning the fasteners on the linked objects.  I don't use Assembly 4 but this fix at least takes care of the issue for my simple case.

luzpaz, 2023-03-06:
@alexneufeld is this ticket still relevant after your fix? If not, what is left ?

alexneufeld, 2023-03-06:
No, this behavior is completely avoided with the current workbench version.

closed 2023-03-06 by luzpaz