shaise/FreeCAD_FastenersWB #93: Error messages when add a PCB metric standoff

branilson, 2021-04-08:
Hello,

I am getting the messages below when added PCB metric standoffs into an assembly in FreeCAD 0.19.

[code]
10:42:28  copying macro: start_bolts.FCMacro
10:48:50  PropertyPythonObject::toString(): failed for <class 'PEMInserts.FSPcbStandOffObject'>
10:48:50  Traceback (most recent call last):
  File "/tmp/.mount_FreeCA5Zji2M/usr/lib/python3.8/json/__init__.py", line 231, in dumps
    return _default_encoder.encode(obj)
  File "/tmp/.mount_FreeCA5Zji2M/usr/lib/python3.8/json/encoder.py", line 199, in encode
    chunks = self.iterencode(o, _one_shot=True)
  File "/tmp/.mount_FreeCA5Zji2M/usr/lib/python3.8/json/encoder.py", line 257, in iterencode
    return _iterencode(o, 0)
  File "/tmp/.mount_FreeCA5Zji2M/usr/lib/python3.8/json/encoder.py", line 179, in default
    raise TypeError(f'Object of type {o.__class__.__name__} '
<class 'TypeError'>: Object of type Quantity is not JSON serializable
[/code]

shaise, 2021-04-09:
can you give a list of steps to reproduce this issue?

branilson, 2021-04-10:
When using A2plus, just adding a PCB Standoff to the assembly that the message appears in the report view box. The file gets corrupted itself if saved with a standoff inserted. We can see it when importing this file as a subassembly in another a2p assembly. Some parts don't show up due to file corruption caused by the presence of the standoff.

If we try to add the same standoff to a part file, we need to press save to see the error message. It looks the issue has something to do with the file save procedure.

In both cases, it is possible to manipulate the standoffs without problems. Therefore I can resize, change the thread and move it to another place. As a workaround, I am converting these PCB standoffs to simple non-parametric parts and removing the parametric ones from the assembly.

shaise, 2021-04-14:
Hi @branilson
Thank you, It should be fixed now. Please test

branilson, 2021-04-16:
Hello @shaise 

Sorry for the delayed response. I wanted to do a detailed test before answer you.

Yes, the error messages are gone. But, persists the issue about the corruption of A2plus subassembly files when a PCB standoff is present. I tried the same with other fasteners besides PCB standoffs and confirmed this issue only happens with this component present.

For your better understanding, I made a simple assembly with a good subassembly (with screws) and a bad subassembly (with PCB standoffs) added. Notice one of the plates of the bad subassembly is missing within the final assembly and when you open the subassembly file directly, everything is there. The files are in this [link](https://drive.google.com/file/d/1noF9gk4YvEmByTThTVm7xgTs-gnV8bue/view?usp=sharing)

Anyway. Your WB is fantastic. It is helping me a lot. Thank you for your support.

luzpaz, 2022-05-31:
@danmiel if you have an inclination, can you also check this (https://github.com/shaise/FreeCAD_FastenersWB/issues/93#issuecomment-821371117) in a2plus?

DanMiel, 2022-06-06:
Update:
luzpaz. I have experimented with the fasteners and have found that ALL PEM style fasteners will create the same problem, which seems to be that the part the PEM is attached to disappears when an assembly is imported into an upper assembly. 
For A2plus: My work around would be to add the fastener to an empty file, add it as a part, then add the fastener file to the assembly with constraints. This allows a person to update the size of the fastener and also updates the locations of the parts when a part file is updated. When fasteners are added directly to the assembly file updating does not update the part locations when part files are updated as I would expect in a constraint type file.
I will continue to try and find why the part disappears as some people would add fasteners in this way.
Dan

luzpaz, 2022-08-27:
Hi @DanMiel, how's it going ?

luzpaz, 2022-08-29:
@kbwbe maybe you have a moment to improve compatibility between a2+ and FastenersWB ?

shaise, 2022-08-29:
@DanMiel ,
PEM fasteners use now the same system as all others. Can you see if they work now with A2plus?

DanMiel, 2022-08-30:
shaise: With minimal testing the PEM fasteners are working.  Putting them on the same system as the others seem to work.
Thank you for your time.
Dan

alexneufeld, 2023-08-20:
should this be closed?

shaise, 2023-08-21:
Yes, It seems the problem is solved.

closed 2023-08-21 by shaise