shaise/FreeCAD_FastenersWB #101: Tool for tapping holes and external threads not scaling

Gesiima, 2021-09-12:
The tools

-Arbitrary length threaded rod for tapping holes (metric).
-Arbitrary length threaded rod for tapping holes (inch).
-Arbitrary length threaded tube for cutting external threads (metric).
-Arbitrary length threaded tube for cutting external threads (inch).

are not effected in resizing when in settings optimized for 3D printing is enabled.
So inner and outer threads needs be scale manually to fit together after printing.
F.e. 20mm thread, nut needs to be 20,2mm and bolt 19,8mm.
It will be great if that resizing values are working with these thread tools too.

zionlion77, 2021-11-11:
I can confirm this issue.
I thought the proper workflow for creating inner threads in bodies would be to use the thread tapping tool in combination with a boolean cut. Yet, this tool doesn't scale with the settings under the Fasteners preferences, making it impossible to get a fine-tuned fit of outer and inner thread.

shaise, 2021-11-13:
This issue was fixed by @alexneufeld, please test.

zionlion77, 2021-11-13:
I tested the changed version and it works like charm. Thanks a lot!

luzpaz, 2022-03-27:
Closed by fc0763e (thanks @alexneufeld !)

closed 2022-03-27 by luzpaz