shaise/FreeCAD_FastenersWB #102: Insert fastener command doesn't work when diameter is changed #242

xoviat, 2021-09-28:
If both the 'fastener' type and 'diameter' is changed, then the insert fastener command fails.

```
09:29:15  At ~\AppData\Roaming\FreeCAD\Mod\fasteners\FSChangeParams.py Line 342
09:29:15  {'sclassname': 'class Base::ValueError', 'sErrMsg': "'#6' is not part of the enumeration", 'sfile': '', 'iline': 0, 'sfunction': '', 'swhat': "'#6' is not part of the enumeration", 'btranslatable': False, 'breported': True}: Inappropriate argument value (of correct type).
```

This may be a fasteners workbench issue? See Zolko-123/FreeCAD_Assembly4#242.

luzpaz, 2022-03-23:
@xoviat can you discuss this on the forum. It gets a lot more traffic/eyeballs than the Fastener issue queue. See https://forum.freecadweb.org/viewtopic.php?f=8&t=11429

luzpaz, 2022-07-19:
@xoviat ping

luzpaz, 2022-08-22:
@berberic2 do you mind weighing-in ?

luzpaz, 2022-08-29:
@shaise ping

shaise, 2022-08-29:
I can't reproduce this issue. Changing both fastener type and diameter seems to work even if I switch between Inch/Metric types. Need more info.

berberic2, 2022-08-31:
> @berberic2 do you mind weighing-in ?
Sorry, can not reproduce…

luzpaz, 2022-08-31:
@xoviat can you please add you full [about](https://wiki.freecad.org/About) info?

alexneufeld, 2024-02-25:
This should be closed, since no one has been able to reproduce the problem.

closed 2024-02-25 by shaise