shaise/FreeCAD_FastenersWB #103: Creating a 1/4" screw is impossible

GhostJumper, 2021-10-20:
When enabling the thread on a 1/4" screw, I get this message

```
16:40:45  Traceback (most recent call last):
  File "C:\Users\Unrea1\AppData\Roaming\FreeCAD\Mod\fasteners\FastenersCmd.py", line 162, in execute
    s = screwMaker.createFastener(fp.type, d, l, threadType, True)
  File "C:\Users\Unrea1\AppData\Roaming\FreeCAD\Mod\fasteners\ScrewMaker.py", line 342, in createFastener
    return self.createScrew(type, diam, len, threadType, shapeOnly)
  File "C:\Users\Unrea1\AppData\Roaming\FreeCAD\Mod\fasteners\screw_maker.py", line 949, in createScrew
    screw = self.makeIso4017_2(ST_text, ND_text,l)
  File "C:\Users\Unrea1\AppData\Roaming\FreeCAD\Mod\fasteners\screw_maker.py", line 1848, in makeIso4017_2
    rthread = self.makeShellthread(dia, P, halfturns, True, offSet)
  File "C:\Users\Unrea1\AppData\Roaming\FreeCAD\Mod\fasteners\screw_maker.py", line 3439, in makeShellthread
    TheShell = Part.Shell(TheFaces)
<class 'Part.OCCError'>: Shape is not a shell
16:40:49  <PropertyLinks> PropertyLinks.cpp(3185): PropertyXLink export without saving the document
```

[image]

OpenCAD Version: 0.19 64-Bit

luzpaz, 2022-08-24:
Is this still relevant ?

alexneufeld, 2022-12-23:
> Is this still relevant ?

No.

luzpaz, 2022-12-23:
Thanks, closing

closed 2022-12-23 by luzpaz