shaise/FreeCAD_FastenersWB #114: Need help implementing JCIS screws

marvin-klein, 2021-11-18:
I want to implement JCIS (Japanese Camera Industry Standard) screws. The standards are found here: [here](https://www.cipa.jp/e/std/std-sec.html) (bottom of the page under CIPA AD-001 to -003). The threads are metric (although it uses 1.4, 1.7, 2.0, 2.3 and 2.6 nominal diameters).
I followed the contributing guidelines as far as I could, meaning writing the tables and entries into the various .py files. The one thing I'm not able to do is actually writing the code. I have zero experience programming a shape for FreeCAD. And because the code in screw_maker.py is not well enough documented so that I could understand what's going on, I'm having issues starting.

I'll attach what I've done so far. I'll include the specsheets and two FC models made according to them.

Regarding my thought process with the non-Phillips-head driver: A "punch" is easy to model by the specsheet, as I've done. I thought of boolean-cutting the recess out of the screws using the "T"-depth from the specsheet. This seems to me to be the easiest way to achieve the recess. The screws themselves seem similar, but not equal to, the ISO7045 to -48 standards.

Again, modeling would not be a prob
lem; understanding and reconstructing the code is. Any help so that I could do it would be appreciated.
[FastenersWB-JCIS.zip](https://github.com/shaise/FreeCAD_FastenersWB/files/7563676/FastenersWB-JCIS.zip)

luzpaz, 2022-02-07:
anyone want to jump in on this?

luzpaz, 2022-03-23:
@marvin-klein I mentioned this FR on the [dedicated Fasteners thread](https://forum.freecadweb.org/viewtopic.php?p=582337#p582337)

BTW, have you looked at https://github.com/shaise/FreeCAD_FastenersWB/blob/master/CONTRIBUTING.md ?

alexneufeld, 2023-12-31:
FWIW, I don't think it makes sense to add fasteners that are specific to only one type of manufactured product to this workbench. DIN, ISO, and ASME standards are widely adopted enough to be worth implementing and maintaining. I think that this is too niche to be relevant to most FastenersWB users.

luzpaz, 2025-02-10:
maybe this could be an optional 3rd party niche option that fasteners can pull into it ?