shaise/FreeCAD_FastenersWB #117: Add support for rivets

mfraser, 2022-01-08:
Is it possible to add support for solid rivets such as NAS1097 & MS20426?

luzpaz, 2022-03-27:
Forum thread discussion: https://forum.freecadweb.org/viewtopic.php?f=8&t=67503

luzpaz, 2022-05-03:
@hpbmo interested in tackling this? :smiley:

hpbmo, 2022-05-04:
@luzpaz 
I don't know very much about rivets and even less about the imperial meassuring system. But I will have a look into some datasheets and then we will see .... .

@mfraser can you describe your needing a little bit more?

mfraser, 2022-05-08:
These are a couple of standards on 2 of the main rivets I use. You would also need to include the formed tail - http://www.pilotfriend.com/experimental/build_12.htm
[nas1097-spec-sheet.pdf](https://github.com/shaise/FreeCAD_FastenersWB/files/8646505/nas1097-spec-sheet.pdf)
[NASM20426-AIA.pdf](https://github.com/shaise/FreeCAD_FastenersWB/files/8646507/NASM20426-AIA.pdf)

mfraser, 2022-05-19:
Before I try to look at this myself, is there any way of creating a non threaded countersunk bolt at the moment?

shaise, 2022-05-21:
by default all fasteners are created without the thread.

hpbmo, 2022-05-23:
@mfraser 

Excuse me for the late response. At the moment I had some other tasks on my desk ;-)

I started with the source files for nas1097 rivets. Maybe I can soon upload them and you can take a look, if they are ok

mfraser, 2022-05-31:
> @mfraser 
> 
> Excuse me for the late response. At the moment I had some other tasks on my desk ;-)
> 
> I started with the source files for nas1097 rivets. Maybe I can soon upload them and you can take a look, if they are ok

Many thanks for doing this. I suppose once one type of rivet has been done, it will be easier to modify it for other solid rivets.

luzpaz, 2022-09-26:
@hpbmo ping  
Any traction yet?

FreeCAD-Tools, 2022-09-27:
How you're going to display the rivets? in theory, they should have two states. riveted and not.

[image]

mfraser, 2022-09-27:
I suppose you'd want them displayed formed if possible.

FreeCAD-Tools, 2022-09-28:
It's just my thoughts.
For example, in fasteners there is a PEN Standoff.
And this fastener has a **blind** parameter.
And if the user changes the blend parameter, the shape of the fastener itself changes.

[image]

It would probably be nice to have such a switch for rivets. to display the formed and unformed state.

hpbmo, 2022-09-30:
@luzpaz 
I will still try to implement the rivets. But  first I want to understand the changes witch shaise had made. I hope to get to that in the next few weeks. 

@FreeCAD-Tools 
That's exactly, what I would try to do. Your hint with parameter **blind** is a good input.

hpbmo, 2022-12-09:
Hi 
I'm still on it, but I can't  spend more time at the moment. 

The files for FsData are nearly done. They have to be in mm or is there way to use the values in inch? In FastenerBase.py I found def LenStr2Num(LenStr). Is this function for direct conversion inch to mm?

Would it be correct to make a new FSFunction FSmakeRivet.py?

I figured out, that I have to add a line in ScreMaker,py and FastenerCmd.py. Where else?

@luzpaz 
Any hints what I missed so far?

shaise, 2022-12-10:
> Would it be correct to make a new FSFunction FSmakeRivet.py?
> 
> I figured out, that I have to add a line in ScreMaker,py and FastenerCmd.py. Where else?

Both are correct. But if you need new type of parameters that do not exist already, it can be a little more complicated.
Regarding Inch values, please look how other Inch fasteners were implemented.

hasecilu, 2024-03-06:
Related #6