shaise/FreeCAD_FastenersWB #118: ScrewTap cannot set to custom.

Thamone, 2022-01-10:
I tried to use ScrewTap but I can only choose from the list(M1.6-M64). If I select Custom from dropdown menu, I will got error message
```

11:26:29  Traceback (most recent call last):
  File "/home/taman/.FreeCAD/Mod/fasteners/FastenersCmd.py", line 486, in execute
    s = screwMaker.makeScrewTap(self.type,d,l,p,d_custom)
  File "/home/taman/.FreeCAD/Mod/fasteners/screw_maker.py", line 3795, in makeScrewTap
    dia = self.smNutThrScaleA * dia + self.smNutThrScaleB
<class 'UnboundLocalError'>: local variable 'dia' referenced before assignment
```
I did manual install but it's still cannot use.

OS: Manjaro Linux (GNOME/gnome)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24366 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0f9259c)
Hash: 0f9259cda103ae1824ac16c68ac9b4a0d54b05fc
Python version: 3.9.7
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.3
Locale: English/United States (en_US)

shaise, 2022-01-10:
Good catch!
Fixed now. Please update.

Thamone, 2022-01-13:
> Good catch! Fixed now. Please update.

Thank you for your support and great Tools.

luzpaz, 2022-05-31:
thanks shaise! closing

closed 2022-05-31 by luzpaz